1technophile/OpenMQTTGateway #1: Duplicate

prahjister, 2017-01-03:
Hey fantastic job on this but I was wondering if you could implement a simple change to remove duplicates. My sensors send duplicate codes to ensure that it is received. Can you implement duplicate removal code? I have changed the code to basically do this but only checks the last code that was sent. This fails if one of the codes sent is not read properly or if 2 sensors fire at the same time. I am using a toggle feature in home assistant so makes it difficult and i will sometimes get 2 toggles.

I used the time feature now() and imported.

1technophile, 2017-01-03:
Thanks, I added the modification into this test program:
https://github.com/1technophile/433toMQTTto433_ESP8266/blob/1technophile-patch-1/_433toMQTTto433_ESP8266.ino

Feel free to test it and comment, if it is ok I will modify also the one with IR.

prahjister, 2017-01-03:
ill test this evening but looking at the code it looks almost exactly like what I implemented. The issue is that it assumes only 1 sensor is transmitting at the same time and that rf is always deciphered correctly. It would be better to log the last 20-30 maybe more in a (vector or array not sure....limited programming experience) and test against all of them. Maybe push each in and the last starts going away after reaches predefined limit.

1technophile, 2017-01-03:
ok, can we sumup your problem like this:
sensor A send 123456 at 14h23m56s12
sensor A send 123456 at 14h23m56s23
sensor B send 654321 at 14h23m56s33
sensor A send 123456 at 14h23m56s35
sensor B send 654321 at 14h23m56s43

currently you are seeing by mqtt:
sensor A send 123456 at 14h23m56s12
sensor A send 123456 at 14h23m56s23
sensor B send 654321 at 14h23m56s33
sensor A send 123456 at 14h23m56s35
sensor B send 654321 at 14h23m56s43

with the last modification you will see by mqtt:
sensor A send 123456 at 14h23m56s12
sensor B send 654321 at 14h23m56s33
sensor A send 123456 at 14h23m56s35
sensor B send 654321 at 14h23m56s43

and you want to see by mqtt:
sensor A send 123456 at 14h23m56s12
sensor B send 654321 at 14h23m56s33

is it correct?

prahjister, 2017-01-03:
yes exactly if the last time is milliseconds

For clarification
i get between 6 and 8 duplicate mqtt requests per trigger within about 2 seconds with an occasional wrong decipher

1technophile, 2017-01-05:
Downloadable for test purposes here:
https://github.com/1technophile/433toMQTTto433_ESP8266/blob/1technophile-patch-1/_433toMQTTto433_ESP8266.ino

prahjister, 2017-01-05:
looks good...will test out tonight....if i wanted to make bigger buffer can I just make a bigger array?

1technophile, 2017-01-05:
you need to change also the number of iteration 10 when this line appears: for (int i = 0; i < 10; i++)

prahjister, 2017-01-06:
I didn't end up changing more than 10. What you have worked perfectly. I did test 3 simultaneous and neither one works. They interfere with each other.

closed 2017-01-06 by 1technophile