1technophile/OpenMQTTGateway #3: Add newremote library to take into account RC with learning mode

1technophile, 2017-01-05:
from https://bitbucket.org/fuzzillogic/433mhzforarduino/overview

prahjister, 2017-01-05:
What will this do?

1technophile, 2017-01-05:
enable more devices to work with the gateway like for example DIO remote control with learning mode

1technophile, 2017-01-29:
begining of the wirk here:
https://github.com/1technophile/NewRemoteSwitch

1technophile, 2017-03-18:
Currently the library licensing is not compatible with MIT license

DenDeze, 2017-07-12:
This would make a lot of people happy in the Netherlands!!!!! could you please re-open?

Fuzzillogic, 2017-07-12:
Commit https://github.com/1technophile/OpenMQTTGateway/commit/7f5a6da8ccb4fdb5c0043b88fea5bcebfa8a002e fixes the issue: OpenMQTTGateway changed to GPLv3, the library is also GPLv3, so they are really compatible.

1technophile, 2017-07-12:
Yes they are now licence compatible, @Dendeze could you confirm that the library as it is receive your signals and is able to replicate them?

DenDeze, 2017-07-12:
@1technophile at this moment i use IMPULS http://www.voordeelmuis.nl/img/gif/1246/1246690.gif

this module codes are received and i can send them.

1technophile, 2017-07-12:
@DenDeze With the newremoteswitch library on an esp8266 ?

DenDeze, 2017-07-12:
Sorry, i misread your question. No i was trying today but could not get it to work.

1technophile, 2017-07-12:
@DenDeze do you have an arduino uno or other so as to test it also?

DenDeze, 2017-07-12:
If you want i could try to run the scripts on an arduino mega (if it still works). normaly i would only use the nodemcu but if you want feedback i can provide you with it.

Could you push me in the right direction?

Is it possible to have the old RF commands and the new programmable units in the same code?

1technophile, 2017-07-12:
What could be interesting is to load last version of @Fuzzillogic from here:
https://bitbucket.org/fuzzillogic/433mhzforarduino/src/0847a6d8a9173abd5abf9cf571a1539f56588c0e/NewRemoteSwitch/examples/ShowReceivedCode/?at=default
and see if it see your codes with your arduino mega

DenDeze, 2017-07-12:
i saw his project today searching for some ways to implement it. i will load the code on the mega and provide you with the feedback. It is possible this will be tomorrow.

DenDeze, 2017-07-12:
Addr 19842046 unit 0 on, period: 269us.
Addr 19842046 unit 0 off, period: 269us.
Addr 19842046 unit 1 on, period: 270us.
Addr 19842046 unit 1 off, period: 270us.
Addr 19842046 unit 1 off, period: 270us.
Addr 19842046 unit 2 on, period: 270us.
Addr 19842046 unit 2 off, period: 270us.
Addr 19842046 unit 2 off, period: 271us.
Addr 19842046 group off, period: 270us.

https://www.50five.nl/apa3-1500r-stopcontact-doos-nl?gclid=CPbIkaDLhNUCFUkQ0wodVVQBrA

There you go!

1technophile, 2017-07-12:
Great thanks, this is positive to extend the gateway possibilities!

DenDeze, 2017-07-12:
@1technophile  thank you for all the time spend in this project. if you can make both protocols to work on the same esp it would be great! (old and new type of hardware).

1technophile, 2017-07-12:
will try, maybe with the help of @Fuzzillogic if he has some time to spend on this

DenDeze, 2017-07-14:
@Fuzzillogic it would be grate if you coupe help with this!

DenDeze, 2017-07-14:
@1technophile for now iT would be grate if in the rf config you coupe specify old op new library

Fuzzillogic, 2017-07-23:
I've switched to z-wave devices. Their two-way communication and encryption makes it more robust. I already had a simple little multi-role home server / htpc, which with a z-wave USB transceiver and Domoticz now gained the extra role of domotica hub. So frankly, I've lost interest in these 433MHz devices.

Also, transmitting using the library should work fine, but receiving and decoding is quite susceptible for noise and the used receiver. With proper math and physics skills that undoubtedly could be improved, unfortunately I lack both in this regard.

minnesberg, 2017-07-31:
Hi.
I am experiencing the exact same issue, meaning that I can receive / send codes to my 433mhz switches using the library "NewRemoteReceiver" the commands that i reviece are similar to what @DenDeze  is describing above.
Addr 22607914 unit 0 on
Addr 22607914 unit 0 off
Meaning it is the same message for all buttons on the remote but then it has additional parameters in unit and state (on / off)
The outlets I am using is Nexa (very common in the nordic countries) and the manufacturer of Tellstick Duo / Net etc.

Is there any plans / progress / roadmaps regarding this ?
It would be nice to be able to use this generic MQTT solution instead of buying the RFXTRX433E adapter.

Your work is much appreated.
Thanks / Ronnie

1technophile, 2017-08-19:
I got it working with a wemos and DIO chacon remote control, which will enable me to integrate it.
As a first approach it will not possible to run RCswitch receiving and newremoteswitch with the same receiver. Regarding the emitter I didn't tested it yet.

DenDeze, 2017-08-22:
@1technophile, maybe it would be possible to just choose Znew Zol 433 in the config? i just tested the lightshow example on an NODEMCU V3 with succes. I only changed pin number to 10 (D2 on the nodemcu). And i removed the dim level part from the example.

1technophile, 2017-08-22:
Yes i will do it like that, i m going to test if we cn keep both for sending.

1technophile, 2017-08-22:
I confirm we can keep both for sending, nevertheless I encounter some stability issue with ESP will see.

DenDeze, 2017-08-23:
If you want i can test to...

1technophile, 2017-08-23:
@DenDeze thanks for proposing, I'm interested. I finish the implementation and go back to you.
Currently I have a working gateway for sending MQTTtoRF2 with this kind of command:
`sudo mosquitto_pub -t home/commands/MQTTtoRF2/CODE_8233267/UNIT_0/PERIOD_272 -m 0`

but it is not receiving yet

1technophile, 2017-08-26:
ZgatewayRF2 published, I didn't implement the receiving of code to the gateway (seems to have conflict with other libraries). So as to get the code of your remotes use ShowReceivedCode from the library below
https://github.com/1technophile/NewRemoteSwitch

Once you get the infos publish the parameters with mqtt like that for off:
`sudo mosquitto_pub -t home/commands/MQTTtoRF2/CODE_8233243/UNIT_0/PERIOD_272 -m 0`
for on:
`sudo mosquitto_pub -t home/commands/MQTTtoRF2/CODE_8233243/UNIT_0/PERIOD_272 -m 1`

Tested OK I reproduce the signal from a DIO PBT-707

closed 2017-08-26 by 1technophile

larsenglund, 2017-09-22:
@minnesberg I can confirm that ZgatewayRF2 works with Nexa switches and dimmers