1technophile/OpenMQTTGateway #15: Add the ability to set the parameters on a web page instead of the code

1technophile, 2017-02-05:


1technophile, 2017-04-08:
some ideas for uno + W5100:
http://forum.arduino.cc/index.php?topic=55044.15

ctripodi, 2017-06-13:
Hi, I agree, could be great to have this option enabled. 
Another example to be considered https://github.com/tzapu/WiFiManager

DenDeze, 2017-07-14:
Check out esp-link for esp devices.

The nodemcu Lua firmware can be flashed with a webpage to set settings.

Must be an option on arduino ide

animavitis, 2017-10-03:
https://github.com/marvinroger/homie-esp8266 have nice GUI to setup wifi/mqtt or custom paramters. Maybe migrate to homie-esp8266?

1technophile, 2017-10-03:
Indeed but i never checked the arduino compatibility do you have some infos about it  ?

animavitis, 2017-10-03:
nope, but i will try to check it

1technophile, 2018-01-19:
duplicate of #83

closed 2018-01-19 by 1technophile