1technophile/OpenMQTTGateway #18: Only uns RFtoMQTTtoRF

outrun0506, 2017-03-01:
Hi,

i can't figure out if it is possible remove the IR Suppprt for the OpenGateway. Unfortunately i can't find something in the wiki if it is possible?

Any Help on that

Thanks

BR

Lukas

1technophile, 2017-03-01:
Hi,

I didn't tested it , but think if you just don't connect the ir devices and upload the program it will work. If not i will add this possibility

outrun0506, 2017-03-01:
Hi, thanks for quick response give it a try tomorrow and come back.

outrun0506, 2017-03-06:
Sorry for late reply, was a little busy last days.

I tried to flash the Sketch to an ESP-8266, only connected to a RF-Receiver and Sender.

I get following Error if i try to flash:

```
C:\Users\lux05_000\Documents\Arduino\libraries\IRremoteESP8266-62c3362e10e324e734a4e54633ac9dbbf865e738\IRremoteESP8266.cpp: In member function 'void IRsend::sendKelvinator(unsigned char*)':

C:\Users\lux05_000\Documents\Arduino\libraries\IRremoteESP8266-62c3362e10e324e734a4e54633ac9dbbf865e738\IRremoteESP8266.cpp:625:14: error: 'KELVINATOR_STATE_LENGTH' was not declared in this scope

   for (; i < KELVINATOR_STATE_LENGTH; i++)

              ^

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).
```

Can you guess where the problem is?

BR

Lukas

1technophile, 2017-03-06:
Hello,

Yes there was an [error ](https://github.com/markszabo/IRremoteESP8266/pull/77)in the IRRemote ESP8266 library that is now corrected in to the master branch, donwload the following code and place it into you library folder:
https://github.com/markszabo/IRremoteESP8266/tree/master

Try to compile and send again the code

outrun0506, 2017-03-06:
Hi @1technophile thanks for help. Not it's working perfect. Also i can confirm the the Gateway is wokring without any IR-Devices installed. So i only have the RF-Sender and Receiver connected an it is working as expected.

1technophile, 2017-03-06:
Hello, thanks for the feedback!
I included[ a page](https://github.com/1technophile/OpenMQTTGateway/wiki/Using-the-gateway-with-RF-only-or-with-IR-only) into the wiki about this

closed 2017-03-06 by 1technophile