1technophile/OpenMQTTGateway #19: Add bluetooth sensor for presence detection

1technophile, 2017-03-04:


wolass, 2017-03-27:
Did you think about including a HM-12/ HM-10 with ibeacon firmware on it? To form an ultimate BRIDGE ? ;)

http://www.instructables.com/id/make-iBeacon/?amp_page=true

1technophile, 2017-03-27:
Thanks for the link, I have an hc-06 that I wanted to integrate but I will take a look at HM-10 and 12

1technophile, 2017-04-12:
hm-10 ordered!

1technophile, 2017-05-01:
Bluetooth LE gateway added :
```
Sending BT data to MQTT
home/BTtoMQTT/7F04B36BA3F2 -057
home/BTtoMQTT/462C2834FFE1 -060
home/BTtoMQTT/CCCCCCCD1A4C -085
home/BTtoMQTT/445F3666113E -056
home/BTtoMQTT/63F9FE29FBC2 -056
BT successfully sent by MQTT
```

Tested on NODEMCU not on arduino uno.

wolass, 2017-05-02:
Impressive!
Did you have to upload any code to HM-10? or it works just by adding the power source to the module?

If you programmed it - could you include the code on the HM-10 side in the wiki?

THANKS FOR KEEPING UP THIS EXTRAORDINARY WORK!

1technophile, 2017-05-02:
Nop didn't upload any code to the Keyes HM 10 , I got it from BG and worked out of the box
https://www.banggood.com/KEYES-HM-10-6-Pin-Transparent-BLE-Bluetooth-V4_0-Serial-Port-Module-With-Logic-Level-Translator-p-1023524.html?rmmds=myorder

Just plug the 4 wires upload the code to the ESP and enjoy.

As I'm not very familiar with beacon use I used Beacon simulator on an android phone to test it.

1technophile, 2017-05-06:
Tested OK on arduino UNO with some minor modifications. I'm closing the request. Don't hesitate to reopen it or open a new issue if needed.

closed 2017-05-06 by 1technophile