1technophile/OpenMQTTGateway #20: Homie convention

oriolrius, 2017-03-07:
Hi, first of all let me congratulate for such good project. I love it and I'm investing quite time on testing what you explain, so thanks a lot.

I just want to ask if you considered on following the `[Homie convention](https://github.com/marvinroger/homie)` on top of the MQTT layer? I used to use OpenHAB and now I'm migrating to Home Assistant and in all devices that I integrate using MQTT I follow that convention, I think it's very useful.

Regards and thanks your opinion about it.

Oriol Rius

1technophile, 2017-03-07:
Hello,

With pleasure,
Yes I considered it but didn't put my hands on it yet. It can be a good point to add some convention into the MQTT world, I will check that.

oriolrius, 2017-03-08:
Nice, thank you for your feedback. I appreciate it.

1technophile, 2017-03-28:
Hello,

What do you think about this convention interpretation

Device: openmqttgateway
Nodes: rf-node, ir-node temperature-node, humidity-node

Properties between ():
openmqttgateway(list of device properties from convention)
    rf-node (433tomqtt(not settable), mqttto433(settable))
    ir-node (irtomqtt(not settable), mqtttoir(settable))
    Temperature-node1 (temperature(not settable))
    Humidity-node1 (humidity(not settable)

433mhz to MQTT would give:
homie/openmqttgateway/rf-node/433tomqtt --> 12344343

MQTT to 433mhz would give:
homie/openmqttgateway/rf-node/mqttto433 --> 12344343

oriolrius, 2017-03-30:
It makes sense, it seems a very good initial approach. Don't forget to report the rest of the variables of the convention is very useful to monitor devices. You have all details here:

https://github.com/marvinroger/homie

It's fast to review.

1technophile, 2017-05-05:
Asked to HA community if homie is compatible with home assistant, waiting for feedback.

animavitis, 2017-10-02:
i have https://github.com/animavitis/mqtt-gateway for ESP8266.
Its based on this great project, but is developed in platformIO IDE & homie, so for sure its possible :)

IoTPlay, 2019-04-15:
Adding another vote for Homie support on this closed issue.... 
Did you look at https://homieiot.github.io ?

1technophile, 2019-04-15:
Issue is still open ;-)

Yes I took a look, and has now OpenHAB support auto discovery with Homie this request is making more and more sense

NorbertHD, 2021-11-08:
Over 2 years have passed since your last comment, is there still hope for the Homie support in the near future?

1technophile, 2021-11-08:
Not sure of the interest on this one as the current discovery is working already with Openhab and hass. Which controller are you using ?

NorbertHD, 2022-01-07:
Thank you @1technophile for the answer. I have a ESP-WROOM-32 controller and want to read Xiaomi Mi Flora Plant Sensors.
But with the uncertainty if and when Homie would be implemented, I thing I use a Raspberry Pi Zero 2 W and ThomDietrich/miflora-mqtt-daemon.
Regards, Norbert

1technophile, 2022-01-07:
@norbert with which software are you going to use the daemon or the gateway ?

NorbertHD, 2022-01-08:
openHAB 3.2

1technophile, 2022-01-08:
Thanks, is OpenHAB + Homie convention discovery proposing more features than OpenHAB + Home Assistant convention discovery ?

NorbertHD, 2022-01-08:
I don't know if Homie or HomeAssistant convention is better. I started with miflora-mqtt-daemon when openHAB and OpenMQTTGateway had not yet HomeAssistant autodiscovery. But miflora-mqtt-daemon runs unreliable and I wanted a low power gateway near the flowers so I looked at OpenMQTTGateway.
I will try/investigate HomeAssistant autodiscovery some time.

Thank you!

xanderificnl, 2022-10-14:
Hi,

I think homie has a lot of advantages. It's self-contained, small, concise and framework agnostic. 

Take this example - formatted as how you'd see it via MQTT-explorer - for instance:
```
homie/
    switchbot-meters/
        $name It's actually a switchbot and less fancy, but a man can dream
        $state ready
        $nodes living-room,kitchen,garage
        living-room/ {kitchen/, garage/}
            $name Living Room
            $properties temperature,humidity,battery,rssi
            temperature/ = 22
                $name Current temperature
                $unit °C
                $datatype integer
            humidity/ = 65
                $name Relative humidity
                $unit %
                $datatype integer
            battery/ = 100
                $name Battery left
                $unit %
                $datatype integer
            rssi/ = -51
                $name Received Signal Strength Indicator
                $unit dBm
                $datatype integer
```

Why this is useful:

- You can write a single script to subscribe to MQTT and have it ingest _all your data_ into influxdb. After all, if there's an update to 'homie/switchbot-meters/living-room/temperature' you already know to set `measurement` to `unit` and `label` to $name. 
- Automated warnings, 'homie/#/$state == lost' 
- Ingesting data becomes standardized. OpenHAB automatically recognizes homie devices and it's sweet!
- It requires no inherit knowledge of the system that's using or providing the data. Unlike HomeAssistant, there are no redirects to `state_t` and `avty_t` -- I don't have to look up any extra information to figure out what the abbreviation means and I don't have to look up which unit belongs to `unit_of_maes` (i.e. units['ssri'] = dBm)

I do feel there's a lot of reasons more why homie is preferred - but most of that comes down to vendorized data. Whenever I've had to deal with HA's MQTT convention, I've ended up spending more time in the documentation that actually solving whatever problem I'm having. I don't actually mean to bash on HA - but I'll admit there's some frustration. 

In short: homie is self contained, concise and provides _all the benefits_ of being clear and standardized.

1technophile, 2022-10-16:
Hello,

I will welcome any PR that implements Homie discovery convention. I can help also pointing to the good parts of the code.

xanderificnl, 2022-10-16:
I'd love to take a crack at it, but I'll admit - I haven't done any (real) work in C/C++ and I've yet to get OpenMQTT working* 

 \* a small aside: the issue is that even with the switchbox right next to it - it only picks it up _sometimes_ - whereas https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32 has no issue. I've been lost on where to start debugging - perhaps you could give me a pointer?

DigiH, 2022-10-16:
Hi @xanderificnl 

> a small aside: the issue is that even with the switchbox right next to it - it only picks it up _sometimes_ - whereas https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32 has no issue. I've been lost on where to start debugging - perhaps you could give me a pointer?

Is it a SwitchBot Bot you have, or any other SwitchBot device? I'm sure the switchbox was just a typo, right?
What exactly do you mean it only picks it up **sometimes**? The changes initiated by another SwitchBot controlling device, only sometimes or only after a while? 

Which OpenMQTTGateway binary do you have installed that you are seeing these problems?
Possibly try the `esp32dev-ble-cont` binary to see if this helps with what you are seeing.

xanderificnl, 2022-10-16:
Hi!

Thanks for the reply. It's a Switchbot Meter Plus, and my first BLE device - and I've wanted a _real excuse_ to setup OpenMQTTGateway for awhile now - so it's the first thing I did.

I assumed there was something I wasn't doing right, so I was fiddling a lot with settings and various options so I'll do my best to recall (it was only yesterday... but I did try a lot of things)

- pressed the button on the meter - short presses didn't seem to work; 
- send a MQTT command to enable SBS1 en later another SMBT (iirc) which I found in your (DigiH's) decoder repository;
- lots of random things and options.

Almost certainly it's a combination of things I did, but I never did get more than one or two temperature / humidity readings. So I'll grab another esp32 and try the binary you're suggesting. I'll open a new issue and tag you with the new results. 

Cheers!

DigiH, 2022-10-16:
> Almost certainly it's a combination of things I did, but I never did get more than one or two temperature / humidity readings. So I'll grab another esp32 and try the binary you're suggesting. I'll open a new issue and tag you with the new results.

For the SwitchBot Meters it should not be necessary to install the `esp32dev-ble-cont` binary, the default `esp32dev-ble` will be fine and should give you a SwitchBot Meter reading about once every minute. 

Please try monitoring your MQTT broker messages with an app like [MQTT-Explorer](https://github.com/thomasnordquist/MQTT-Explorer/releases/tag/v0.3.5), from which you can also paste some messages here if the problem persists.

> so I was fiddling a lot with settings and various options so I'll do my best to recall (it was only yesterday... but I did try a lot of things)

That could have caused some of the problems of seeing less messages than you actually are expecting. Are you using Platformio to make all the changes and fiddling? ;)

If so, and if worst comes to worst, you can always re-download the fresh source and start from scratch.

Hope this helps.

1technophile, 2022-10-23:
Regarding the Homie convention, I think we should have a kind of separate Python script that detects the "gateways" and apply either Homie, either Home Assistant convention.
There are a lot of programs that publish sensor data as a gateway:
* amr2mqtt
* rtl_433
* OMG
* ESPurna
* ...

Instead of replicating the code into each of these, the python script could push HA or Homie discovery following user preferences.
This script could be added as an Addon into HA or a binding into OpenHAB.

RTL_433 as such implementation for HA, https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433_mqtt_autodiscovery
but I think it could be extended to other projects and conventions.

xanderificnl, 2022-10-23:
I've been thinking about the approach used by the remote decoder. Simply have OMG publish raw data to MQTT and have a script smash and squish the data into the homie format. It'd be relatively easy to write another adapter to export into the HA-format. FWIW, OpenHAB natively supports homie.

Is that also what you're suggesting, but as a gateway agnostic approach? I'd be curious of something like [Telegraf](https://www.influxdata.com/time-series-platform/telegraf/) could be used.

1technophile, 2022-10-24:
Yes, I think that works well. 
To sum up:
OMG -> gateway driver - converter - Homie convention driver -> OpenHAB
Theengs -> gateway driver - converter - Homie convention driver -> OpenHAB

I haven't used Telegraf so challenging to say.

1technophile, 2023-06-12:
Closing this, if anyone wants to open a PR feel free.

closed 2023-06-12 by 1technophile