1technophile/OpenMQTTGateway #22: other ethernet module/library support

evilgreen, 2017-03-13:
Wiki says

> W5100 shield (tested) or **other** ethernet module

I have enc28j60 ethernet module and trying to adapt the sketch to use UIPEthernet library. Hope I'm doing things right...
So, I'm only getting the following:
```
client mqtt not connected, trying to connect
Attempting MQTT connection...

```

Will it be possible to provide the sketch for such hardware, even untested? I will be more than happy to test and share results. Thanks!

evilgreen, 2017-03-14:
It does work with enc28j60 and Mega 2560, I had to adjust the Ethernet library to work with Mega.

closed 2017-03-14 by evilgreen

1technophile, 2017-03-14:
hello could you please give me the adjustement you've made to work with the mega?

evilgreen, 2017-03-14:
Actually, I was about to ask your advise as well. I fixed the ethernet part and even got a few messages from RF, but now I see nothing. My hardware should be ok as it was tested with RFLink before. I have only RF receiver connected, no IR at all.
So, for ENC28J60 on Mega 2560 the following adjustments needs to be done:
1. in {libraries}\UIPEthernet\utility\Enc28J60Network.h :

> //UIPEthernet(SPI_MOSI, SPI_MISO, SPI_SCK, SPI_CS);
>   #if defined(TARGET_LPC1768)
>     #define SPI_MOSI p51 ; p11
>     #define SPI_MISO p50 ; p12
>     #define SPI_SCK  p52 ; p13
>     #define SPI_CS   p53 ; p8

2. ENC28J60 needs to be wired to above pins (50-53)

@1technophile could you please advise if any pin and/or timer needs to be changed for Mega?

1technophile, 2017-03-14:
Pin D3 (#define RF_RECEIVER_PIN 1) should be ok on the mega, this board has an [interrupt ](https://www.arduino.cc/en/Reference/AttachInterrupt)on this pin.
What did you change between the receiption you've got and now?

evilgreen, 2017-03-14:
That was very quick check in the morning before going to work... Now everything is back as it was initially but I've got only 1 message which probably caused by some noise:

> home/sensors/ir 365390630

Note I have no IR receiver!

Earlier it was another message which is probably more relevant

> home/433toMQTT 7701071

but again it was only single message

Question - do I need to follow the following note?

> //IMPORTANT NOTE: On arduino UNO connect IR emitter pin to D9 , comment #define IR_USE_TIMER2 and uncomment #define IR_USE_TIMER1 on library <library>IRremote/IRremoteInt.h so as to free pin D3 for RF RECEIVER PIN

BTW, the file name is wrong, it should be boarddefs.h

1technophile, 2017-03-14:
Yes you need to follow this note, or change your receiver to pin 18 (#define RF_RECEIVER_PIN 5) or another free interrupt pin

evilgreen, 2017-03-15:
Looks like a library issue. The examples provided with rc-switch library are not working either.