1technophile/OpenMQTTGateway #26: Byron

scargill, 2017-03-20:
Hi

I'm very new to this having been using the RFLink433 unit -  someone pointed out this code - so - I put together an ESP8266 and the RF receiver...   I am looking at the serial signal and with a little handset I have - the unit is picking it up no problem (other than reboots sometimes when setting up).

However it is totally ignoring my Byron SX35 doorbell pushes - should it know about these?

1technophile, 2017-03-21:
Hi,

When you get the code of this doorbell into rflink do you have the protocol  name given, if yes could you share it?

scargill, 2017-03-21:
"20;91;Byron;ID=00f9;SWITCH=01;CMD=ON;CHIME=01; "

1technophile, 2017-03-21:
I don't think it's a protocol supported by rcswitch, after a few search it seems to be an homeeasy protocol that could be supported by the newremoteswitch library:
https://github.com/1technophile/NewRemoteSwitch

I would like to integrate it to the gateway but the licence is currently not compatible with MIT one, waiting for the developper answer about this.

scargill, 2017-03-21:
I thought I’d give that a shot – out of the box with an ESP8266. Could not get anything out of it.

 

https://github.com/1technophile/NewRemoteSwitch

1technophile, 2017-03-22:
Thanks for testing this. It's a pity that It doesn't work.

If you have an arduino and some time to test here is some track to follow:
https://bitbucket.org/fuzzillogic/433mhzforarduino/wiki/Home
https://github.com/bruce33/homeeasyhacking

If the test are concluant I would relaunch the dev for changing their licence and try to integrate and port the code for the gateway.

scargill, 2017-03-22:
The first one uses the same libraries as earlier…

 

Second one no longer works with current Arduino!

 

https://bitbucket.org/fuzzillogic/433mhzforarduino/wiki/Home
https://github.com/bruce33/homeeasyhacking

1technophile, 2017-03-22:
If I sum up:
newremoteswitch on esp8266 ko
remoteswitch with arduino ko
new remote switch with arduino ko
homeeasyhacking with arduino ko

Is that correct?

scargill, 2017-03-22:
Bear in mind I’ve not gone into this deeply but one says it does not work on Arduino 1 – the other uses the two libraries used in the original ESP package – and I had no luck with that. I can’t really say any more than that. The RFLink software I do know works. Best way is to grab the software and try it.

1technophile, 2017-03-22:
Yes I understood that your bell works with RFlink seeing you first serial extract nevertheless I can't integrate it as it is not anymore open source.

DenDeze, 2017-07-12:
Dear scargill, since i am a node-red user i know your name from your blog.

1technophile has by far made the best MQTT to RF code for the esp (i use the nodemcu V1).

i just e-mailed the maker of the newremoteswitch code hoping for a reply so that 1technophile could make his project even better.

My old receivers work liek a charm with his code (controlled by node-red).

FOr the self-learning kind of switches support would make the project score 200%.

closed 2017-07-23 by 1technophile