1technophile/OpenMQTTGateway #28: Cannot compile with error  "avr/interrupt.h: No such file or directory"

rexkani, 2017-03-24:
Dear all,

I'm very new to this arduino environment and this is actually my first project to explore my smart home study. I have installed the Platformio software, imported the "OpenMQTTGateway" project folder and compiled the code. i quickly realised that i am also required to get all the related libraries and put into the lib folder to make use of the additional libraries. 

i have got myselfe IRremote, IRremoteESP8266, pubsubclient, and rcswitch from the internet and put into the lib folder. my compiling process went further until i met the following situation which i cant proceed anymore:

`Compiling .pioenvs\nodemcuv2\lib\Ethernet\utility\w5100.o
Compiling .pioenvs\nodemcuv2\lib\rc-switch-master\RCSwitch.o
Compiling .pioenvs\nodemcuv2\lib\Arduino-IRremote-master\IRremote.o
lib\Arduino-IRremote-master\IRremote.cpp:21:27: fatal error: avr/interrupt.h: No such file or directory
#include <avr/interrupt.h>
^
compilation terminated.

*** [.pioenvs\nodemcuv2\lib\Arduino-IRremote-master\IRremote.o] Error 1
 [ERROR] Took 17.75 seconds`

i realised this is a standard lib from the arduino suite and i went to install the arduino software, copied all the lib and put into Platformio global lib but i still cant get past this error.

Can anybody please help to enlighten me on this? thank you very much in advance.

Regards,
Kenny

1technophile, 2017-03-24:
hello Kenny,

I didn't tested the code with platform.io.
Especialy I don't know if it takes into account the code compilation options. In particular the compilator should not take arduino libraries if you choose nodemcu board which seems to be the case in your logs.

I recommend to use Arduino IDE 1.8.1 or higher with board manager esp8266 2.3.0 or higher which is tested OK

I don't close the issue as it can be interesting to make the code compatible with platform.io but if you want to progress now with the gateway I recommend you the arduino IDE.

rexkani, 2017-03-25:
Hi 1technophile,

Thanks for replying. I have spent more time studying on that, and i found that the missing libraries are actually standard libraries with arduino. and its included in the basic development kit in both arduino and Platformio. i just dont know why it cannot be found by Platformio. 

as you said im only using esp8266, i realised i dont need the arduino codes so that i commented out the arduino part like this, is this the right way to do?

`#include "User_config.h"
#include <PubSubClient.h>
#include <RCSwitch.h> // library for controling Radio frequency switch

// array to store previous received RFs, IRs codes and their timestamps
unsigned long ReceivedSignal[10][2] ={{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0}};
/*------------------------------------------------------------------------*/

//adding this to bypass the problem of the arduino builder issue 50
void callback(char*topic, byte* payload,unsigned int length);

RCSwitch mySwitch = RCSwitch();

#ifdef ESP8266
  #include <IRremoteESP8266.h>
  #include <ESP8266WiFi.h>
  IRrecv irrecv(IR_RECEIVER_PIN);
  IRsend irsend(IR_EMITTER_PIN);
  WiFiClient eClient;
//#else
  //#include <IRremote.h>
  //#include <Ethernet.h>
  //EthernetClient eClient;
  //IRrecv irrecv(IR_RECEIVER_PIN);
  //IRsend irsend; //connect IR emitter pin to D9 on arduino, you need to comment #define IR_USE_TIMER2 and uncomment #define IR_USE_TIMER1 on library IRremote.h so as to free pin D3 for RF RECEIVER PIN
#endif`

a little side track question, can we specify static IP for ESP8266, and add additional SSID for it?

1technophile, 2017-03-25:
Hello,

With the latest version of arduino IDE you should not have to do that, but what you've made is correct

Yes we can specify static IP for ESP8266, I will add it to the code

rexkani, 2017-03-26:
thank you so much!

but i guess the secondary SSID option is not possible?

1technophile, 2017-03-26:
It is working for you now?
you mean to connect to a second SSID if first one is not available?

rexkani, 2017-03-27:
haha i dunno if its working right now.

for the secondary SSID, i was just seeking for an option.

the wifi successfully connected, and when i use MQTT to try the publish RF, it shows on the serial monitor that it tried to send and received an ACK. But i can never get back a received signal from 433toMQTT. now im actually wondering if i have ever got the transmitter and the receiver interacted with the nodemcu at all. i'm now stuck here and dont know how to debug and proof.

[image]
[image]
[image]

1technophile, 2017-03-27:
Some questions, did you get the msg on the serial saying that MQTT is connected?
What are the devices that you are trying to command if you are trying to send and what are the devices that emits if you are trying to receive?

rexkani, 2017-03-27:
MQTT was connected, i saw it from the mosquitto.log that the nodemcu did connect to it.

i was trying to mimic the command of some "cadja" branded remote switches from china.

i was trying to read the command from the remote control, but when i didnt get any feedback, i tried to emit command by myself using the sample code in the wiki. but i didnt receive anything either. i wonder if i did really emit anything, or received anything from the emitter/receiver

1technophile, 2017-03-27:
If you open the serial monitor did you see message MQTT message received by the gateway when you publish it by mosquitto?
And do you see some traces into the serial monitor when pressing your remote control buttons?

rexkani, 2017-03-28:
If you open the serial monitor did you see message MQTT message received by the gateway when you publish it by mosquitto?  
///YES
And do you see some traces into the serial monitor when pressing your remote control buttons?
///NO

1technophile, 2017-03-28:
Did you check your wirings and voltages?
If you have a multimeter you could verify that you have 5V for your emitter and receiver power supply

rexkani, 2017-03-28:
i guess i need to find a multimeter first.

[image]

1technophile, 2017-03-28:
Yes, and check the receiver part first, do you have a link showing the device that you want to control?

1technophile, 2017-03-28:
Nevertheless, I propose that you open a subject into OpenHAB forum 
https://community.openhab.org/s
or contact me by PM on this forum. I'm going to close the issue here as we are far from the first error.

closed 2017-03-28 by 1technophile