1technophile/OpenMQTTGateway #32: [question] nRF905 SUPPORT?

wolass, 2017-03-27:
Hi, your work is AWSOME! But did you consider adding other RF transmitters? 
Like for example nRF905 transceiver? 
I could not find anything about connecting it to a Esp8266, but maybe you know more about this

1technophile, 2017-03-27:
hello, 

Thanks! I was thinking about adding nrf24l01 but not this one, haven't heard about it. I have seen that there is some arduino libraries but didn't found also ESP8266 support

consolacion, 2017-04-12:
You  made a great program, I guess adding anything and still keep one file to maintain is going to be a bit tricky as for the UNO you only have some 576 bytes left for local variables. That is abt 50 bytes away from the dreaded   'unstable' warning.
So yes, adding any functionality would be great, but i guess you can only do that for the ESP8266 compilation ;-)
So,,, while we are at it, how about RFM69 support ;-)

1technophile, 2017-04-12:
Thanks, my strategy is as long as I can maintain one program I will do it. It has the advantage of making it optimal.
I'm more inclined about supporting RFM69 support as it seems more used than nRF905.

RFM69 seems to me dedicated to DIY or some special board like jeenode or moteino. What is the use case you are seeing with RFM69?

consolacion, 2017-04-12:
Wise decision to keep it in one file.
I was joking a bit about the RFM69 :-) Not sure if RFM69 is more dedicated to DIY than  any other electronics project out there. There are various boards, except the ones you mention Adafruit has 2  featherboards with RFM69 I think.
I use it  for longRange connection of sensors that are some 300 meters away in another property that has no internet. 
Am using the setup of Computourist (github.com/computourist) who has done a great job, but ofcourse if one can kep it all in one Gateway... even better (see caveat later).
Other than the RFM69, can opt for an RFM95 and make 'true' LoRa connections (meaning following the right  protocol and being able to connect to 'the things network'

Now let's not get overboard and add  stuff purely for the sake of adding, For any serious LongRange communication, further than for instance your mailbox at the end of yr driveway, it is probably likely you need to find a proper place for the gateway to create a LineOfSight. Such a place may not be a problem for the 433MHz connection, but it may not be the right place for your IR LineOf Sight.

As such, I think your initial Idea of adding NRF24L01 support is much better than RFM69 or RFM95

1technophile, 2017-04-15:
If we take into account the range i think rfm69 or 95 are better than nrf24l01. But nrf24l01 is more used especialy with mysensors framework...

consolacion, 2017-04-15:
I fully agree. I was just  thinking abt the practicalities of combining RFMxx (which usually needs to be placed high and outside) and the IR (which is more aimed at inside the house),
But I would welcome any addition, either RFM69 or NRF24l01

1technophile, 2017-04-15:
The more I think and the more the RFM69 seems to be a good next step, I have also an open energy monitor that rely on RFM69 and will enable me to simplify my architecture a little.

consolacion, 2017-04-15:
I am looking forward toit :-)

1technophile, 2017-04-22:
I have ordered a set of RFM69, will be my next play!

consolacion, 2017-04-22:
Looking forward to that.
I had an RFM69 on an arduino pro mini (3V3)  but transferred it to a Wemos. Separate though from the IR/RF wemos.  Would be interesting to see it all in one

wolass, 2017-07-11:
How is the tinkering with RFM69 going?
I actually am very interested how this works out as this would greatly increase the capabilities of your great all-in-one GATEWAY.
Mostly by including ENCRYPTION of the communication between NODES and gateway AND ACKNOWLEDGEMENTS - as new nodes with RFM69 have receive/transmit possibilities. 

The library is more or less ready from the gihub/lowpowerlabs/RFM69

The other project that I found which provides RFM69 to MQTT communication is the 
https://github.com/bbx10/nanohab

Would this be useful for your project?

Let me know what you think

1technophile, 2017-07-11:
Hello, thanks for the links. I m planning to work on it this month. There is also the gateway from @computourist on which i will take a look at

1technophile, 2017-08-02:
First step I have now two rfm69hcw connected to esp8266 communicating ! (not so easy), remains the code which I don't think will be a major issue

closed 2017-08-26 by 1technophile

1technophile, 2018-03-24:
RFM69 is already implemented, Lora is also on the roadmap

knopserl, 2018-03-25:
Hey great news, LoRa is perfect!!! I  was already thinking to switch to the TTN LoraWAN/LMIC project, but if LoRa is on the (hopefully) near roadmap for OpenMQTTGateway, I prefer to stay "here".
I was thinking to start my own attempts to use the RadioHeas RFM95/LoRa lib's but can save the effort and wait if it's done in the roadmap of this project. I love this LoRa boards especially as they have all onboard with and ESP WiFI chips which is perfect for a gateway (small, cheap, no soldering required)
Thanks!

knopserl, 2018-03-25:
my initial thought was lora node <--> OMG <-> MQTT (OpenHAB or Mosquitto/OpenHAB). If I can directly  integrate OMG with OpenHAB I will of course skip Mosquitto. 
For the lora node, I would use the same devices/PCB as for the OMG such as the Heltec ESP32 + SX1276/RFM95 and SHT31 (as the best TempHum sesnor) and maybe a BME280 for air pressure and an e-Paper for display the 3 values. Thus my additional request for SHT31 and e-Paper support ;-).
Another quick question: does/can the OMG pass the date/time to the lora node so show it also on the e-paper? As typically sensors show the date/time?
thanks and regards from Austria!

1technophile, 2018-03-31:
> Another quick question: does/can the OMG pass the date/time to the lora node so show it also on the e-paper? As typically sensors show the date/time?
I cannot answer for now, not digged enough yet but I will keep it in mind.
Have a good evening