1technophile/OpenMQTTGateway #33: Could you add support for AC's maybe, like Mitsubishi

1technophile, 2017-03-27:
Awesome project !!!
Although i can't get 433ghz receiver to snif anything, IR works fine.
Could you add support for AC's maybe, like Mitsubishi.
ex: https://github.com/markszabo/IRremoteESP8266

thank you for work, keep it up :D !

PetricaM, 2018-01-30:
Hi Florian,

I don't think it is feasible to implement support for AC protocols as they are simply too many and require quite advanced knowledge for average user. Moreover, the support should be implemented in IRRemote in order for OpenMQTT to recognize the signal.

However, there's an workaround. I think that it should work for most of the AC.

1. Simply record all the button codes send by the remote as raw signals; here IRRemote has an issue as for AC signals are much longer than for regular IR sent by TV/VCR as it includes the full state of the device (target temperature, mode, fan, swing, etc); you actually told me what line to modify in IRRemote as it was recording only about 400 char (I think that it was IRRecv.h line #define RAWBUF       100U  // Length of raw duration buffer but I don't I recall exactly :) ); when you release the pre-compiled versions of OpenMQTT you can modify the respective IRRecv.h line to have the full signal recorded.

The buttons on my remote are On/Off; Sleep; Fan; Swing; Mode; Temp. As a result, there are 153 different signals sent by the remote.

2.  Create a script with the payload for each button press (153 scripts :) ). The name of the script is important as it includes the button pressed. 
livinghvacoff:
  alias: off
  sequence:
    - service: mqtt.publish
      data:
        topic: "home/GatewayOpenMQTT7/commands/IR_Raw"
        payload: "3106,1704,526,..."
livinghvaconnoautofalsecool16:
  alias: onnoautoFALSEcool16
  sequence:
    - service: mqtt.publish
      data:
        topic: "home/GatewayOpenMQTT7/commands/IR_Raw"
        payload: "3052,1702,528,..."
3. Set component MQTT HVAC; the topics where the component posts the commands are being used in step 4 to be concatenated and matched against scripts names:
- platform: mqtt
  name: HVAC Living
  modes:
    - dry
    - cool
    - fan
  swing_modes:
    - yes
    - no
  fan_modes:
    - auto
    - high
    - medium
    - low
  hold:
    - yes
    - no
  power_command_topic: "hvac/living/power/set"
  mode_command_topic: "hvac/living/mode/set"
  temperature_command_topic: "hvac/living/temperature/set"
  fan_mode_command_topic: "hvac/living/fan/set"
  swing_mode_command_topic: "hvac/living/swing/set"
  hold_command_topic: "hvac/living/swing/set"
  current_temperature_topic: "hvac/living/state/temperature"
  payload_on: "on"
  payload_off: "off"
  retain: true
  initial: 22

4. Sensors; the first 6 sensors are intermediary and are based on the topics sent by the HVAC component for each of the remote buttons; the next 6 are used for dealing with the error messages after a reboot (5% core code, 95% error handling :) ).

- platform: mqtt  
  state_topic: "hvac/living/power/set"
  name: 'HVACi living power'  
- platform: mqtt  
  state_topic: "hvac/living/mode/set"
  name: 'HVACi living mode'  
- platform: mqtt  
  state_topic: "hvac/living/temperature/set"
  name: 'HVACi living temperature'  
- platform: mqtt  
  state_topic: "hvac/living/fan/set"
  name: 'HVACi living fan'
- platform: mqtt  
[... 87 lines cut ...]
  action:
    - service: homeassistant.turn_on
      entity_id: script.hallhvacoff
- alias: 'HVAC hall sleep mode off'
  initial_state: 'on'
  trigger:
    - platform: sun
      event: sunrise
      offset: "+02:00:00"
  action:
    - service: mqtt.publish
      data:
        topic: 'hvac/hall/sleep/set'
        payload: "no"
- alias: 'HVAC hall sleep mode on'
  initial_state: 'on'
  trigger:
    - platform: sun
      event: sunset
      offset: "+02:00:00"
  action:
    - service: mqtt.publish
      data:
        topic: 'hvac/hall/sleep/set'
        payload: "yes"

waraileron, 2018-07-01:
Hi there, this is my temporary workaround for my mitsubishi ac, activates on mqtt topic IR_AIRCON with 0 being off and anything above 10 being on. (I do not see the need to change other parameters but could be a point to start looking into it if you do need them.)
1. add in config_IR.h
`  #define IR_AIRCON `
2. add in ZgatewayIR.ino
`  #include <ir_Mitsubishi.h>`
3. then add this bit of code in ZgatewayIR.ino above #ifdef ESP8266  // IR_RCMM not available on arduino
```//test aircon code here  
  #ifdef IR_AIRCON
  if (strstr(topicOri, "IR_AIRCON") != NULL){
    IRMitsubishiAC mitsubir(IR_EMITTER_PIN);
    mitsubir.begin();
    if (data == 0){
    mitsubir.off();
    mitsubir.setFan(0);
    mitsubir.setMode(MITSUBISHI_AC_COOL);
    mitsubir.setTemp(data);
    mitsubir.setVane(MITSUBISHI_AC_VANE_AUTO);}
    else if (data > 10){
    mitsubir.on();
    mitsubir.setFan(0);
    mitsubir.setMode(MITSUBISHI_AC_COOL);
    mitsubir.setTemp(data);
    mitsubir.setVane(MITSUBISHI_AC_VANE_AUTO);}
#ifdef ESP8266
  mitsubir.send();
#endif
    signalSent = true;
  }
  #endif
//end test code
```

1technophile, 2018-07-28:
[Added](https://github.com/1technophile/OpenMQTTGateway/commit/b12ee1997954d4c6968600664fe0c5aab0cbbac3)

1technophile, 2018-07-28:
@PetricaM I took the opportunity to add your idea to the [wiki](../IR-Air-climatiser-Home-Assistant-integration) thanks

closed 2018-07-28 by 1technophile