1technophile/OpenMQTTGateway #35: Arduino Duemilanove woes

IronWolfMC, 2017-04-17:
I've been attempting to use the gateway with an Arduino Duemilanove using a 433mhz RF transmitter/receiver, yet have been having quite a fair amount of difficulty in getting it to work. It seems that whenever I send an RF command, my RF transmitter will only send gibberish to the receiver. This transmitter/receiver pair has been working fine with my Raspberry Pi, so the transmitters do indeed work. The transmitter is connected to pin D4.

The console debug log: https://gist.github.com/IronWolfMC/c0afb79a549ab70cf66458b97755f1ae

RF Receiver output log (the code I sent was 5330227, instead the receiver is getting a single random digit): https://gist.github.com/IronWolfMC/271948abb980b2efcd95b47f6ce2437f

The Arduino is also receiving 'phantom IR transmissions' based on the console log, of which I cannot figure out what is going on. I do not have a receiver connected.

Thank you for any help.

1technophile, 2017-04-17:
Hello,

I see the following potential issue:
-is your arduino based on atmega328p or atmega168 (should be written on the chip), if atmega168 i think there is not enough ram to support.
-secondly did you modify the irremote library following this note:
**IMPORTANT NOTE: On arduino UNO comment the following line ** #define IR_USE_TIMER2 line 137 and uncomment #define IR_USE_TIMER1 line 138 into the library IRremote/boarddefs.h so as to free pin D3 for RF_RECEIVER_PIN

IronWolfMC, 2017-04-17:
The board is based on the Atmega328 (not p version though). The board is pretty much exactly the same as an Uno.

I did indeed comment/uncomment the necessary lines in the IRremote library. The transmitters I am using are these: http://a.co/9mg98Av which from what I understand will work perfectly fine.

1technophile, 2017-04-17:
When you compile did you get the warning message about lack of memory for variables ?

IronWolfMC, 2017-04-17:
Nope. No error, compiles just fine.

`Sketch uses 28478 bytes (92%) of program storage space. Maximum is 30720 bytes.
Global variables use 1510 bytes (73%) of dynamic memory, leaving 538 bytes for local variables. Maximum is 2048 bytes.`

1technophile, 2017-04-17:
Could you provide me with irremote, rcswitch, arduino ide and openmqttgateway version?

IronWolfMC, 2017-04-17:
Sure thing.

OpenMQTTGateway: Latest @ 97eaf15
IRremote: 2.2.1 (also tried @ 048efb2)
RCswitch: @ f820d2e
Arduino IDE: 1.8.1

I should have tried the v0.2 release of OpenMQTTGateway, I'll give that one a try this afternoon.

1technophile, 2017-04-17:
Don't think 0.2 will change something, could you indicate to which pin your receiver is connected. Did you checked the voltage supplied to the emitter ?

IronWolfMC, 2017-04-17:
In my latest tests, I haven't had an receiver connected, nor any IR components. All I had connected is the RF transmitter to pin D4 (which is powered by a regulated 5v power supply and has been tested with it to work with the RPi sending the codes before). I did connect the 433 receiver to D3 on the Arduino to see if it would receive codes sent by other devices, and it did not report them to MQTT.

1technophile, 2017-04-17:
You mean that just with one rf emitter connected you see ir signals appearing in the traces ?

IronWolfMC, 2017-04-17:
Yeah. That's what I find rather odd. Nothing but the RF transmitter connected to D4.

1technophile, 2017-04-17:
Wow that's pretty strange, if you remove the transmitter do you have the same behaviour ?

IronWolfMC, 2017-04-17:
I believe so. I'll check again to confirm it shortly.

1technophile, 2017-04-17:
Are you using w5100 or another ethernet module?

IronWolfMC, 2017-04-17:
Standard [W5100](http://a.co/4Nw6Ey8). Network connectivity works fine.

IronWolfMC, 2017-04-17:
Confirmed, it does happen even with nothing connected and not sending any commands via MQTT. It just magically picks up some IR commands. https://gist.github.com/IronWolfMC/b9a906a3f31884455711834c775b2071

IronWolfMC, 2017-04-17:
Hmm, I tested it also with an ESP8266 and it also shoots gibberish to the receiver as well.

1technophile, 2017-04-18:
I will retest on my platforms and go back to you

1technophile, 2017-04-18:
I'm trying to reproduce your issue with the same versions of library as you nevertheless, I can't find an IRremote library with version 2.2.1
https://github.com/z3t0/Arduino-IRremote/releases
https://github.com/markszabo/IRremoteESP8266/releases

IronWolfMC, 2017-04-18:
Strange. I removed the library version I had and reinstalled release 2.1.0 of IRremote, however my Arduino IDE reports the version of IRremote to be 2.01 (when it's 2.1.0).

Just for extra safety, I also replaced the rcswitch library with 2.6.2 and it made no difference.

Interestingly enough, when I added an extra '#include IRremoteInt.h' on line 57 of the main script, I only got it to receive a single 'ghost' transmission.

Without it on line 57 (multiple IR signals received, nothing connected): https://gist.github.com/IronWolfMC/6d2b7eb1f5d2dd87d5972f9c8267d91b

1technophile, 2017-04-18:
could you upload to your arduino the rcswitch senddemo.ino example and see if you receive something with the pi?

IronWolfMC, 2017-04-19:
Yeah, looks like it sends gibberish as well. https://gist.github.com/IronWolfMC/9b2f1461b5bbe102fbce71f9c0c54ed5

I'm starting to wonder if the RF transmitter is not compatible with the rcswitch library, but it still makes me wonder why there would be a 'ghost' IR transmission.

1technophile, 2017-04-19:
Yes there is something strange with this ghost IR could you try the gateway with this code instead, so as to exclude the last commits I've made:
https://github.com/1technophile/OpenMQTTGateway/tree/568c8f0d3076d1ed71c4d0b4312dd09a2323d90d

Regarding the RF issue, do you have other way to receive data than the pi? To double check the receiver.
You could try also to send the signal with different pulselength and protocols:
mosquitto_pub -t home/commands/PLSL_315/433_4 -m 1315156 will make RCswitch use the protocol 4 with a pulselength of 315ms
I would began by testing different pulselength first

Do you know what RF library your pi is using?

IronWolfMC, 2017-04-19:
The [RF receive script](https://raw.githubusercontent.com/milaq/rpi-rf/master/scripts/rpi-rf_receive) uses the python rpi_rf package. I've been testing these against my 433mhz RF outlets as well by sending their activation codes which do not respond to the command. The receiver is known good as it does receive the proper commands that come out of the 433mhz remote that came with the outlets.

I did try playing around with the pulselength option a bit but it did not seem to change much.

Also, when testing the senddemo.ino file, it appears to have potentially made my ethernet shield stop working unfortunately. I may have to see if there's something that can be done to revive that in the meantime.

1technophile, 2017-04-19:
I'm suspecting a conflict between the ethernet shield and the interrupt used by the libraries, could you try senddemo without the ethernet shield?

IronWolfMC, 2017-04-19:
Still the same gibberish even without the shield.

1technophile, 2017-04-19:
you mean that you are receiving only one digit of signal data?

IronWolfMC, 2017-04-19:
Yup. Usually a single digit of digital signal with a random pulselength of anywhere from 400 to 900.

1technophile, 2017-04-19:
we are close to the minimal circuit... and I don't see why it is not working. The last possibilities I see:
-load the sendemo example on another board like your esp8266 and see if you have the same behaviour as above
or
-put your esp8266 + receiver with the receivedemoadvanced example, connected to serial and check what is coming as signal from your arduino + transmitter
or
-try with another transmitter like an fs1000a (even if I don't believe on this;
, due to the fact I have also a superheterodyne one looking like yours plugged to my arduino uno)

IronWolfMC, 2017-04-22:
It looks like it was the power supply having a hissy-fit over the Arduino or ESP being connected, as the same PSU works fine with the RPi + transmitter. Tested with a separate PSU and it works great with ESP.

Thank you so much for your help. Is there any way I can buy you a coffee for helping me with all these antics?

1technophile, 2017-04-22:
I'm happy you figured it out! 
Thanks for proposing, I will accept with pleasure a tea :-), here is an email adress for paypal transfer untechnophile@gmail.com
or if you are planning to come to west of France I will be happy to share it in real.

closed 2017-04-22 by 1technophile