1technophile/OpenMQTTGateway #38: Problems with Sony IR

phlash13, 2017-04-30:
I had to change the following code in "ZGatewayIR.ino" from
`  if (strstr(topicOri, "IR_Sony") != NULL){
    irsend.sendSony(data, 12);
    signalSent = true;`

to

`  if (strstr(topicOri, "IR_Sony") != NULL){
    irsend.sendSony(data, 15, 3);
    signalSent = true;`

Because my Sony AVR relies on having IR signals repeated 3 times (this is the last parameter...) and uses 15bit signals...

Maybe you could generalize the function a bit more, or is there a way to tell how many repetitions and which bit-length to use via MQTT?

1technophile, 2017-04-30:
there is not at this time, but it's a good idea I will add it

wolass, 2017-05-02:
This actually is an issue in the IRremote library rather than openMQTTGateway see this issue: https://github.com/markszabo/IRremoteESP8266/issues/124

phlash13, 2017-05-02:
This _was_ an issue with IRremote, and as you can see from my example it works with the additional parameter (which isn't used by openMQTTGateway...)!
In addition to that openMQTTGateway uses fixed bit lengths which also isn't that great...

wolass, 2017-05-02:
U R right @phlash13.

1technophile, 2017-05-05:
Added:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-IR#send-data-by-mqtt-with-advanced-ir-parameters

Waiting for your feedback

1technophile, 2017-05-20:
@phlash13 could you close the issue if it is OK for you

closed 2017-05-20 by phlash13