1technophile/OpenMQTTGateway #40: Use status led for checking mqtt connection

1technophile, 2017-05-04:
https://community.home-assistant.io/t/433mhz-infrared-ir-to-and-from-mqtt-on-esp8266/6779/187?u=1technophile

8666, 2017-11-20:
I have a mod that has led for sending, receiving and disconnected from mqtt broker

Landrash, 2017-11-20:
@1technophile If there's room for it in the memory using a WS2812 for different states would be nice. 
The http://fastled.io/ library is rather good.

1technophile, 2017-11-20:
@8666 it could be interesting to submit a pull request with this mod
@Landrash didnt know this driver, interesting, maybe you could open an issue as an enhancement diffenrenciated from this one. As this one was more for gateway state and the ws2812 seems more for fairy lights

8666, 2017-12-03:
I made pull request but I have this "The Travis CI build failed"
https://github.com/8666/OpenMQTTGateway

1technophile, 2018-07-28:
already integrated

closed 2018-07-28 by 1technophile