1technophile/OpenMQTTGateway #43: Integration of 315 Mhz RF

Rudy96, 2017-05-06:
It would be nice to integrate RF-Receiver and RF-Tranmitter for 315 Mhz.
Using Modules für 315 Mhz with the actual settings for 433 Mhz does not work.

1technophile, 2017-05-06:
Could you detail what is not working exactly, you are not able to receive, send or both?

Rudy96, 2017-05-07:
I have only on hardware with 315 Mhz and I don't know the codes to send, so I can't test the transmitter before i get the values by reciving them. So I tested only the Receiver by pressing the buttons of the original remote. But nothing happend on the mqtt-broker (i monitored it with mqtt.fx). With the IR-Receiver everything is fine and I can see the messages on the mqtt-broker.

I use this Receiver and Transmitter from ebay: http://www.ebay.de/itm/315Mhz-RF-transmitter-and-receiver-Wireless-link-kit-for-Arduino-Raspberry-Pi-/272450136713?hash=item3f6f4b2a89

1technophile, 2017-05-07:
It could be also that your remote control emit with a non supported protocol.

Could you try to emit with an arduino and a [send example](https://github.com/sui77/rc-switch/blob/f820d2e46ab227ac800960b57cdefc8f912006c6/examples/SendDemo/SendDemo.ino) of rcswitch to see if the gateway receive it.

Rudy96, 2017-05-08:
Thanks for your help. I tried the "send example" with an Arduino nano. But I can't receive any data with the OpenMQTTGateway. If I understand it right, with the "send example" there should or has to be a message in mqtt.fx. I have no idea what is wrong, but I am also no professional with Arduino, Wemos and things like that.

1technophile, 2017-05-08:
In this case it's more a question of a non supported protocol by rcswitch library than a 315mhz incompatibility.

Rudy96, 2017-05-09:
Ok, thanks for the support. Then I will try to solve my problem at the rcswitch project. So this issue can be closed.

closed 2017-05-09 by 1technophile