1technophile/OpenMQTTGateway #44: Turn on and off RF trasmitter

Guglio95, 2017-05-08:
Hi, 
Is the 433MHZ Tx always transmitting carrier signal even if i'm not trasmitting data?
Is it better to connect TX VCC to a GPIO and turn it on/off? 
ESP8266 GPIOs carry 12mA, that should be ok for powering the TX unit.

1technophile, 2017-05-09:
Hi,

I'm doing that on atmega328p weather stations that run on batteries to avoid residual consumption. The drawback on doing that on the esp is the transmitter powering with 3,3v, it will work but the range will not be very good. Best thing maybe is to put a transistor and take the power from 5v or superior power supply.

1technophile, 2017-05-12:
@Guglio95  what is the issue of transmitting carrier signal?

1technophile, 2017-07-12:
@Guglio95 as I don't have answer I close the issue, if you have more details don't hesitate to reopen it

closed 2017-07-12 by 1technophile