1technophile/OpenMQTTGateway #45: Openhab users warning

glasruit, 2017-05-12:
It took me three weeks to find the solution why the IR send function did not work in my Openhab2 configuration. The space character between "{mqtt=">[broker:" and "home/commands/IR_......" is essential.
No space char = not working...

1technophile, 2017-05-12:
strange, I'm not reproducing this behaviour, here is a definition from my item file:
`Switch IRtest				"IR"				<light>	(GF_JA,Plugsm)					{mqtt=">[broker:home/commands/IR_GC:command:ON:38000,1,1,171,171,21,64,21,21,21,21,21,21,21,21,21,21,21,21,21,64,21,64,21,21,21,21,21,21,21,21,21,21,21,21,21,64,21,21,21,21,21,21,21,64,21,21,21,21,21,21,21,21,21,64,21,64,21,64,21,21,21,64,21,64,21,64,21,64,21,1114]"}`

glasruit, 2017-05-13:
@1technophile, Strange indeed. I edit the items file with notepad++ in UTF-8 modus.
`Switch TV_key_On	"aan" { mqtt=">[oh2mqtt: home/commands/IR_GC:command:*:36000,1,1,32,32,64,64,64,32,32,32,32,64,64,32,32,64,32,32,64,32,32,3272]" }`
Without the space character the last trace message is "Hey I got a callback" and nothing happens.

closed 2017-05-13 by glasruit