1technophile/OpenMQTTGateway #46: Standing Fan IR Code drives me Crazy

synologix, 2017-05-16:
First of all, thank you for doing this great project.

I am new with all this stuff and started with Arduino 2 Weeks ago... let a led blinking and thinked.. ok that was easy so look for some cool stuff and here we go....

I think i made all as you have described including "#define MQTT_MAX_PACKET_SIZE 256"
all codes working fine expect the RAW Code....

I cannot find a Global Caché Code for my Fan 
so i have started make my own small code for detecting and sending the codes and endet up with this...

#include <IRremote.h>

IRsend irsend;

// Power On/Off
unsigned int raw2[100] = {8950,4400,600,1650,600,600,550,600,550,550,600,550,550,600,550,600,550,600,550,550,600,1700,550,1650,600,1650,600,1650,600,1650,600,1650,600,1700,550,1700,550,1700,550,600,550,600,550,600,550,550,600,550,550,600,550,600,550,600,550,1650,600,1650,600,1650,600,1650,600,1700,550,1700,550,1700,600,1650,600,550,550,600,550,600,550,600,550,550,600,550,600,550,550,600,550,1650,600,1650,600,1700,550,1700,550,1700,600,1650,600,};

//Speed 
unsigned int raw3[100] = {8950,4450,550,1700,550,550,600,600,550,600,550,550,600,550,550,600,550,550,600,550,600,1650,600,1650,600,1650,600,1700,550,1700,550,1700,550,1700,600,1650,600,1650,600,1650,600,550,600,550,600,550,550,550,600,550,600,600,550,550,600,550,550,1700,550,1700,550,1700,550,1700,600,1650,600,1650,600,1650,600,1650,600,600,550,550,600,550,550,600,550,600,550,550,600,600,550,550,600,1650,600,1650,600,1650,600,1650,600,1650,600,};

// Rotate
unsigned int raw4[100] = {8950,4450,550,1700,600,550,550,600,550,600,550,600,550,550,600,550,550,600,550,600,550,1650,600,1650,600,1650,600,1650,600,1700,550,1700,550,1700,550,1700,600,1650,600,550,550,600,550,1650,600,600,550,600,550,550,600,550,550,600,550,1700,550,1700,550,600,550,1650,600,1650,600,1700,550,1700,550,1700,550,600,550,600,550,1650,600,600,550,550,600,550,550,600,550,600,550,1650,600,1650,600,600,550,1650,600,1650,600,1650,600,};

void setup()
{
  Serial.begin(9600);
}

void loop() {
  if (Serial.read() != -1) {  
  	for (int i = 0; i < 1; i++) {
  		irsend.sendRaw(raw2,100,38); // Power On/Off
  		delay(1000);
      irsend.sendRaw(raw3,100,38); // Speed
      delay(1000);
      irsend.sendRaw(raw4,100,38); // Rotate
      delay(1000);
  	}
  }
}

Works fine... only Problem, its not a Global Caché Code.... 
So i have look around and found a Program called:  IrScrutinizer / HARCToolbox this is  able to convert different Codes to other format 

http://www.harctoolbox.org
https://github.com/bengtmartensson/harctoolboxbundle/releases

and endet up with this:
IrScrutinizer captured signal: <no decode>
0000 006D 0032 0000 0154 00A7 0016 003F 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 003F 0016 0002

+8941 -4391 +579 -1657 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -579 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -1657 +579 -53

00 00 2D 00 D2 04 11 76 08 93 01 21 03 3C 01 21 01 21 01 21 00 1A 32 01 22 22 22 22 11 11 11 11 12 22 22 22 21 11 11 11 12 22 22 22 21 11 11 13

sendir,1:1,1,38029,1,101,340,167,22,63,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,63,22,63,22,63,22,63,22,63,22,63,22,2

so the last entry should it be... so i cutted out to:

mosquitto_pub  -t home/commands/IR_GC -m "38029,1,101,340,167,22,63,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,63,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,22,63,22,63,22,63,22,63,22,63,22,63,22,2"

but nothing happens...

looks like there is something wrong... the LED on the NodeMCU is not flashing when send this..
instead when i use: 

mosquitto_pub -t home/commands/IR_NEC -m 551494365
for my LG TV, the Led is flashing at send

hope anybody can help me what i doing wrong...

Best Regards 
Hannes

1technophile, 2017-05-16:
Hello,

I have added a prototype of sendraw, you can test it with the last commit and with the following instructions:

1) change the mqtt packet size to 512, [this site](https://mothereff.in/byte-counter) enable to count the byte of the list you are passing to the gateway 

2) Enable IR_Raw for arduino by uncommenting the line 129 in user_config.h
` #define IR_Raw`
If you are using the uno you will have to comment other gateway like ZgatewayBT(l60) or ZgatewayIR(l61) to keep enough memory

3) publish your code like below
`mosquitto_pub -t home/commands/IR_Raw -m "8950,4400,600,1650,600,600,550,600,550,550,600,550,550,600,550,600,550,600,550,550,600,1700,550,1650,600,1650,600,1650,600,1650,600,1650,600,1700,550,1700,550,1700,550,600,550,600,550,600,550,550,600,550,550,600,550,600,550,600,550,1650,600,1650,600,1650,600,1650,600,1700,550,1700,550,1700,600,1650,600,550,550,600,550,600,550,600,550,550,600,550,600,550,550,600,550,1650,600,1650,600,1700,550,1700,550,1700,600,1650,600"`

synologix, 2017-05-16:
WOW this was really fast, i am impressed.

Where is the Exorbitant big Smiley in this Comment System? Because it works :-)

Thanks a lot

closed 2017-05-16 by synologix