1technophile/OpenMQTTGateway #47: Adding DHT sensor

BertrumUK, 2017-05-17:
I have enabled the sensor in the User_config.h "#define ZsensorDHT" but I cannot see where the pins are enabled for the dht11 to be picked up from?

BertrumUK, 2017-05-17:
RF is working a treat in HA but I want to remove the wiring from my Raspberry PI 3 and run them from a NodeMCU via MQTT.

1technophile, 2017-05-17:
Data pin for DHT is defined into ZsensorDHT.ino line 30, for nodemcu you can choose maybe GPIO15 which correspond to D8 by writing
`DHT dht(15,DHT11);`

After you connect DHT Vcc to 5V power source and the ground to the ground power source

BertrumUK, 2017-05-17:
Thanks for that. Will changing to DHT22 allow that dht version to work?

1technophile, 2017-05-17:
yes just replace DHT11 by 
`DHT dht(15,DHT22);`

BertrumUK, 2017-05-17:
Thank you very much.

closed 2017-05-17 by BertrumUK