1technophile/OpenMQTTGateway #48: DHT Enable

synologix, 2017-05-17:
Hi,

i have tried to Compile with the option use of DHT but i run in some problems...

first Missing the DHT.h so i installed the "DHT_sensor_libray" from Adafruit
then there was missing the Adafruit_Sensor.h so i installed "Adafruit_Unified_Sensor"

But now i am Ending in: 

ZsensorDHT:38: error: redefinition of 'long unsigned int time1'
 unsigned long time1 = 0;
               ^
ZgatewayBT:36: error: 'long unsigned int time1' previously defined here
 unsigned long time1 = 0;
               ^
exit status 1
redefinition of 'long unsigned int time1'

Probally using the Adafruit Libs is wrong and i must Install other ones ?

Sorry i have started a short time ago with all the Arduino things...
And dont have such a good knowlegde in debugging scripts 

At all i think it would be a good Idea to Expand the instructions for Adding the DHT libs like it was described for the iremote and pubsubclient ;-)

Best Regards
Hannes

synologix, 2017-05-17:
Hmn its Possible i found a Solution for this.. but i dont know this is the right way to do this...

in "ZgatewayBT" i have replaced all  type site /reference
time1 -> timebt 

in "ZsensorDHT"  i have replaced all  type site /reference
time1 -> timedht 

Compile Error is gone but is that the right way?

And sorry for my bad English...

Best Regards
Hannes

synologix, 2017-05-18:
Ok after all i found out this is the right way... I had doubts because issues with the whole system when the DHT22 was connected at boot time with pin d8 

when i boot up the device and plug in then the dht22 on pin8 all works finde

then i found out... no its not my fault at all its a generic problem....
GPIO0, GPIO2, and GPIO15 are boot select pins and this causes the NodeMCU to halt on boot 
see also: http://www.forward.com.au/pfod/ESP8266/GPIOpins/index.html

so i have decided doing the following:
in "ZsensorDHT"  
DHT dht(4,DHT22); // NodeMCU Users should change this to "DHT dht(4,DHT22);" with pin D2 for Wiring

and in "User_config.h"
#define RF_EMITTER_PIN 15 //put 15 = D8 on nodemcu, 4 = D4 on arduino

Recompiled and Uploaded...

Tadaaa Issues gone....

BertrumUK, 2017-05-19:
Hello Hannes,
Thank you for that update - makes sense now why I couldn't flash my NodeMCU when I had a DHT11 attached to D8. I will experiment with another port on the NodeMCU as a DHT input to save any hassle.

As this is resolved you may want to close it  :thumbsup:

1technophile, 2017-05-19:
I tested with a DHT22 connected to SD3 on nodemcu and it worked, by using this pin you are not forced to use D2 (already use by RF emitter)

1technophile, 2017-05-20:
@synologix could you close the issue if it is OK for you

synologix, 2017-05-20:
Sorry nope works not for me... all components work normal expect the DHT22 it measures nothing
no value comes up on mqtt message

so i have tested arround and changed to: DHT dht(3,DHT22); 
this works on the fly ....

possible my ESP8266 is a another model then yours?
i use a "wemos.cc  LoLin NodeMCU V3" and in the Arduino Editor i have choosen NodeMCU 1.0 (ESP-12E Module) 
also Possible this works not because whats described here: 
https://github.com/esp8266/Arduino/issues/1446

i have not started dive deepe into this, because the really good weather today in germany ;-)

1technophile, 2017-05-20:
ok I'm using a nodemcu geekcreet maybe it is the difference. 
As my goal is to have parameters working with most of the boards I have tested with another free pin (it begin to be difficult :-)), the pin D3

`DHT dht(D3,DHT22); //on nodeMCU this is D3 GPIO0`

And it is working. Could you test if it works with your config?

BertrumUK, 2017-05-20:
I've got it working on D3 on a NodeMCU 😉

In the testing phase at the moment using a DHT11, will move over to a DHT22 once I finish my work shift. 

Using this setup under Home Assistant

- platform: mqtt
  name: "MQTT RF Temperature"
  state_topic: "home/433toMQTT/dht1/temp"
  unit_of_measurement: "°C"
  value_template: '{{ value | round(2)  }}'

- platform: mqtt
  name: "MQTT RF Humidity"
  state_topic: "home/433toMQTT/dht1/hum"
  unit_of_measurement: "%"
  value_template: '{{ value | round(2)  }}'

Thanks Florian 👍

1technophile, 2017-05-24:
@synologix is this parameter working for your board:
`DHT dht(D3,DHT22); //on nodeMCU this is D3 GPIO0`

synologix, 2017-05-27:
Yes, working for me, sorry for long time response, i am working on a addon for OpenMQTTGateway ;-) Will hopefully finished soon

1technophile, 2017-05-27:
@synologix great, i close the issue. 
Don't hesitate to submit a pull request I will be happy to review it!

closed 2017-05-27 by 1technophile