1technophile/OpenMQTTGateway #49: json payload

1technophile, 2017-05-20:
instead of passing parameters into the MQTT subject use json

marksev1, 2017-05-23:
It might be also cool if it was definable to use either way? (the old or the new way)

1technophile, 2017-05-23:
yes good idea at short term, at mid term I will remove the old way.
This way the transition can be progressive. Thanks for the idea!

jumpalottahigh, 2017-06-09:
What do you think about the ArduinoJSON library? In my tests it has worked nicely. Maybe we can consider using it for the JSON payload?
https://github.com/bblanchon/ArduinoJson/

1technophile, 2017-06-09:
Yes I think it is a good path

posicat, 2018-01-18:
I just got my IR detection working, and saw that I get 4 MQTT messages back for each time I press the button.  I would definitely prefer to get the data back as one JSON message with fields for each of the parameters.  My main concerns is that I want to have multiple IR gateways, and if 2 devices report at the same time it's harder to determine which go together.

posicat, 2018-03-26:
It seems like we could have 2 .cpp files, one that would create an MQTT message for each return parameter, and another that would accumulate a JSON message for each of those parameters.

ReporterJSON report // Uncomment only one to enable that reporting method
#ReporterMQTT report

For example if we added a device for light sensing that returned RGB values, it could call:
   report.channel("LightSense");
   report.value("Red",128);
   report.value("Green ",200);
   report.value("Blue ",59);
   report.send();

if we include the MQTT reporter library it would send back:
   home/LighSense/Red  128
   home/LightSense/Green 200
   home/LightSense/Blue 59

If we included the JSON reporter library it would instead return
   home/json/  {"LightSense":"{"Red":128,"Green":200,"Blue":59}} 

This might help reduce a lot of redundant code across the project, and allow people to add new sensors/features to the software more readily (They just hook into the same reporting method to send back data)

posicat, 2018-03-26:
I'd love to help with the refactor, I'll take a look at the code and the @brokema code and see if there's something I can add to it.

1technophile, 2018-03-27:
great, don't hesitate to submit a pull request for one gateway for example.
The goal would be to handle all the mechanism that are proper to a gateway on this object

1technophile, 2018-04-21:
@posicat Hello,

I'm asking myself if I wait for json to be implemented to release v0.7, what is your opinion?

posicat, 2018-04-26:
Life's gotten busy lately, I've had relatives moving, and house projects I've had to get to first.  I do want to work on the JSON code but I'm not sure I'll have time to do that anytime soon.

1technophile, 2018-09-24:
Hello,

I'm going to work on this for info

1technophile, 2018-11-05:
#298 done

1technophile, 2018-11-05:
to deactivate jsonpublishing:
comment [this line](https://github.com/1technophile/OpenMQTTGateway/blob/d8dba15a8151f8270b852b89d07b90626aff28da/User_config.h#L143)

to deactivatesimple publishing:
comment [this line](https://github.com/1technophile/OpenMQTTGateway/blob/d8dba15a8151f8270b852b89d07b90626aff28da/User_config.h#L146)

closed 2018-11-05 by 1technophile

rikki78, 2019-02-20:
Could you give an example of the json payload that should be send to the gateway?

1technophile, 2019-02-20:
Yes, here they are:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF-09#send-data-by-mqtt-to-convert-it-on-rf-signal

rikki78, 2019-02-20:
ok thanks. Can you give an example of the parameters to pass for protocol 2 (Rf2)

1technophile, 2019-02-20:
It is also in the wiki:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF2-09#send-data-by-mqtt-to-convert-it-on-rf-signal

rikki78, 2019-02-20:
ah I'm sorry for missing that... I just checked the parts with json in the title