1technophile/OpenMQTTGateway #53: esp8266 enable sendPanasonic and sendRCMM

glasruit, 2017-06-06:
You might want to include IR support for IR_PANASONIC and IR_RCMM. I managed both with a little update of your code.
Inside user_config.h after line 127 add
```
  #define IR_PANASONIC
  #define IR_RCMM
```
Inside ZgatewayIR.ino after line 240 add:
```
  #ifdef IR_PANASONIC
  if (strstr(topicOri, "IR_PANASONIC") != NULL){
    irsend.sendPanasonic(0x4004, data);
    signalSent = true;
  }
  #endif
  #ifdef IR_RCMM
  if (strstr(topicOri, "IR_RCMM") != NULL){
    if (valueBITS == 0) valueBITS = 32;
    irsend.sendRCMM(data, valueBITS);
    signalSent = true;
  }
  #endif
```

I also managed a "long power on" press for the Panasonic, however that needs some additional changes in IRremoteESP8266.h and IRremoteESP8266.cpp

1technophile, 2017-06-06:
Hello,

Thanks for the proposal,  could you post the modification on irremoteesp8266, i will check it and could include it in the wiki.

1technophile, 2017-06-06:
added to master @glasruit please close if ok for you

glasruit, 2017-06-07:
Actually I reported my changes on Github/IRremoteESSP8266. But it seems that these changes are already incorporated in the 2.0-dev branch.  In case that you keep using the original branch, these are my alternations.

My Panasonic tv needs a long press, about 2 seconds, power-on. I managed that, by repeating the power-on-signal for four times with in between a sleeping period of 100 ms.

Change in IRremoteESP8266.h line 203:
```
void sendPanasonic(unsigned int address, unsigned long data);
```
to
```
void sendPanasonic(unsigned int address, unsigned long data, unsigned int repeat=0);
```
And in IRremoteESP8266.cpp line 373:
```
void ICACHE_FLASH_ATTR IRsend::sendPanasonic(unsigned int address, unsigned long data) {
    // Set IR carrier frequency
    enableIROut(37);
    // Header`
    mark(PANASONIC_HDR_MARK);`
    space(PANASONIC_HDR_SPACE);`
    // Address (16 bits)`
    sendData(PANASONIC_BIT_MARK, `
        PANASONIC_ONE_SPACE,  `
        PANASONIC_BIT_MARK,  `
        PANASONIC_ZERO_SPACE, address, 16, true); `
    // Data (32 bits) `
    sendData(PANASONIC_BIT_MARK, PANASONIC_ONE_SPACE,`
         PANASONIC_BIT_MARK, PANASONIC_ZERO_SPACE,`
        data, 32, true);`
     // Footer`
     mark(PANASONIC_BIT_MARK);`
     ledOff();`
 }
```
to
```
void ICACHE_FLASH_ATTR IRsend::sendPanasonic(unsigned int address, unsigned long data, unsigned int repeat) {
    for (unsigned int i = 0; i <= repeat; i++) {
        // Set IR carrier frequency
        enableIROut(37);
        // Header
        mark(PANASONIC_HDR_MARK);
        space(PANASONIC_HDR_SPACE);
        // Address (16 bits)
        sendData(PANASONIC_BIT_MARK, PANASONIC_ONE_SPACE,
            PANASONIC_BIT_MARK, PANASONIC_ZERO_SPACE,
            address, 16, true);
        // Data (32 bits)
        sendData(PANASONIC_BIT_MARK, PANASONIC_ONE_SPACE,
            PANASONIC_BIT_MARK, PANASONIC_ZERO_SPACE,
            data, 32, true);
        // Footer
        mark(PANASONIC_BIT_MARK);
        ledOff();
        delay(100);
      }
}
```
In that case your code should also be updated.
In ZgatewayIR.ino line 243 change
```
irsend.sendPanasonic(PanasonicAddress, data);
```
to
```
irsend.sendPanasonic(PanasonicAddress, data, valueRPT);
```
Finally the Panasonic-long-press-power-on is send with:
```
mosquitto_pub -t home/commands/IR_PANASONIC/RPT_3 -m 16825533
```

1technophile, 2017-06-07:
ok as I'm planning to integrate the 2.0 we will suppose it covers the need, thanks for the infos

closed 2017-06-07 by 1technophile