1technophile/OpenMQTTGateway #57: Light Sensor Addon is finished

HannesDi, 2017-06-07:
i hope all of you understand my bad english ;-)

sorry for the chaos and unstructured work but this is the first contribution 
for a project in my life... i started 3 month ago with the arduino stuff
and have no clue i have understand the git system correct how to merge code..

So here we go....

https://github.com/HannesDi/OpenMQTTGateway/tree/addon-lightsensor

I have opened the Door on OpenMQTTGateway for "i2c Bus Modules/Sensors"
but this has required to complete restructure the Pin Connections
on the NodeMCU

Why?
2 Reasons

First:
As most of us know and and 1technophile has allready written
we run out of usable pins for different boards.
So i have made some research and found out that many pins
could switched its function...
like d0 - gpio16 and d5 - gpio14
d0 could not be used for receive because it has no interrupt as i have understand
so why not use it for Sending IR ? ;-)
the d5 is free and usable for other stuff.

Second:
To Use I2C there a 2 possible ways...
1. complete re-invent the i2c software for the Board 
2. use the Wire.h that exists and work great...
But Wire.h has a small problem.. it explicit wants D1 and D2 on the ESP8266
and so i have freeded and switched the other pins..

Ended up in this:

D0 - GPIO16 # as Send Pin for IR (because IR need no interrupt)
D1 - GPIO5  # as SCL (I2C Bus) for BH1750 Light Sensor
D2 - GPIO4  # as SDA (I2C Bus) for BH1750 Light Sensor
D3 - GPIO00 # as Receive Pin for 433mhz
D5 - GPIO14 # as Receive Pin for DHT22
D8 - GPIO15 # as Send Pin for 433mhz

And all is Working ;-)

There is a another benefit for using this...
I await the next 2 Days a another i2c bus module the "BME280" a "I2C or SPI Temperature, Humidity and Pressure Sensor"

for all they dont know what this means...
the I2C bus can handle up to 8 modules on 2 wires in parallel (no more worries about free pins)
and... for thoose they have allready a DHT11 or DHT22 you can unplug it then and
we have again pin D5 free for other things ;-)

i hope you an first of all 1technophile like my work

Best Regards Synologix / HannesDi

HannesDi, 2017-06-08:
Good News, today my Bosch BME280 multisensor has arrived and i have implented the code to 80% 
Unfortunately today was Thursday and so i could not finish it yet but on friday/saturday i send it out as pull request ;-)

1technophile, 2017-06-09:
Hello, 

Travis enable to do automatic compilation of the code each time there is a change to the repo.  By the way it verify that the code available is correct on a compilation point view. This is a part of a broader domain called continuous integration CI.
Regarding the PR we have to modify it until travis is OK for all boards. You can find the details in the travis logs. The current error is due to the fact that you uncommented the zaddonsensor which i m not testing by CI. You could comment it to check if your PR is OK maybe you will have also to add a library download into the travis.yml file.

HannesDi, 2017-06-09:
ok i found the problem....

and i think i have adopted it from your main branch ;-)

/home/travis/build/1technophile/OpenMQTTGateway/ZsensorDHT.ino:31:17: fatal error: DHT.h: No such file or directory

difference in mine and your User_config.h is:

yours: //#define ZsensorDHT
mine: #define ZsensorDHT

so as mine has bind ZsensorDHT it runs into the missing entry for the dht sensnor lib and adafruit sensors

possible it helps adding:

git clone https://github.com/adafruit/DHT-sensor-library
git clone https://github.com/adafruit/Adafruit_Sensor

to "travis.yml"

for both of us ?

i could test this theorie when i arrive home or maybe you are faster and patch it in

HannesDi, 2017-06-10:
Light Sensor Addon is Obsolete now because replaced by I2C Modules Addon -> So... Close

closed 2017-06-10 by HannesDi