1technophile/OpenMQTTGateway #65: Add information on how to print a PCB to the Wiki

jumpalottahigh, 2017-06-20:
Has anyone created a gerber file for this project? I love the project and I'm using it for 433 MHz devices to HA, but thinking to move away from breadboard and cables into a PCB and a box. I would appreciate any tips or ideas on the subject. Mainly looking at itead.cc and fritzing but they are quite new to me and I was unable to find any out of the box design files. Any good ideas on printing a board for this project or would a point to point prototype board do the trick also?

1technophile, 2017-06-23:
Hello @jumpalotthigh,

I don't have seen yet some pcb examples, for the moment i m using prototype board and it works well.

jumpalottahigh, 2017-06-23:
Hi @1technophile , thanks for sharing. Would you care to post an image of your setup for inspiration?

In my case this is my breadboard setup that works and I'm looking into moving to a PCB:
[image]

And this is my first unsuccessful attempt at soldering on the PCB. I think I shorted something out though. Would it be a good idea to add an image of properly done prototype board set up to the Wiki and close this ticket?

[image]

[image]

1technophile, 2017-06-23:
I don't think my prototype board is interesting, it's made as an arduino UNO shield with some addons (ALARM, 2 status led, 2 DHT) and I'm not very proud about the soldering :-(

Regarding my ESP8266 breadboard setup here is the picture:
[image]

HannesDi, 2017-06-28:
Probally in the Summer Holidays i will start diving into this...
Because i also want to have this on a real PCB

@jumpalottahigh: Do you have the possibility to etch a pcb yourself? 
I ask, because i will try to do make a gerber file but i have no expirience in that and so its possible i / we need some tries to get a working one ;-)

jumpalottahigh, 2017-06-28:
@HannesDi No, I have never done it myself and have very limited experience with circuit making in general. I was thinking if someone with more experience would create the design, you can order professionally made prototype circuits from [Itead](https://www.itead.cc/)

jumpalottahigh, 2018-01-12:
That's awesome! If you do make a prototype I'd totally love to order a few.

posicat, 2018-01-15:
I got my gateway working with IR last night, so now that I have a working circuit I can start moving that onto a PCB design.  It's not going to be a 2-day project, because it's going to take me a little bit to get all the hardware to test how things work.

HannesDi, 2018-01-16:
I was ill for a long Time... Its fine to see that some other also planning do a PCB :-)

i can now slowly start again with doing some stuff.

Time will show when i can show up my release

posicat, 2018-01-18:
Feel free to look over my shoulder on the EasyEDA project I'm working on.  I've built PCBs before but I just do it as a hobby, any advice, especially on missing resistors/capacitors/etc would be great.

posicat, 2018-01-23:
I started a project for IR only modules based on ESP-01 boards.  It's s starting point for a larger board based on the ESP-12F.  I ordered 10 of the boards, and enough components to build all 10.  Eventually I hope to offer this as a kit to people (more likely the ESP-12F version which will have RF, I2C and more) but feel free dance on the bleeding edge and order some of these for yourself if you want to help me test them.  The system has a built in list of materials you can order along side the modules.  Warning, these have not yet been tested, I might (probably have) reversed the connections on something and it will require electronics expertise to get it working, if it all.

https://easyeda.com/posicat/OMG_ESP_01_PCB-445e573315b543108581500a4d4974be

1technophile, 2018-01-23:
@posicat regarding the ESP12F version what do you think about building a first with RF and I2C, and building the others with the "shield" principle, this would look like:

- Base gateway with RF and I2C light, temp, humidity

- IR gateway shield

- RFM69 gateway shield

Please note that I'm not sure we should use D3 for the RF receiver pin as I have many feedback pointing that it is working only with serial monitor ON.

posicat, 2018-01-23:
I think we can fit a lot of the parts onto one board, the design I was working on so far has IR, RF, and I2C and plenty of pins for a DHT22 and a light sensor.  I don't see anything wrong with the shield idea, other than it increases cost to add on more functionality.  It's not much more work to design 2 boards, one all-in-one and another that will take shields (Arduino-sized or something specific to this project)

1technophile, 2018-01-23:
The goal is to adjust the needs to the user, but ones will say to just not solder or add the components related to the gateway he doesn't want.
 Just an idea

1technophile, 2018-01-23:
Please keep in mind that future gateways could come later, I'm thinking to LORA for example

posicat, 2018-01-23:
Right, there will always be new technologies to add to a project like this, and there will always be too few pins to connect all of them.  ESP chips are absurdly cheap, I could see a series of boards that had as many connections as possible on each.  One board I already put together for myself is just IR and I2C (in theory at least, not tested yet) another board could be RF, IR, Temp/Humidity, and Light.

As new technologies come along new boards could be designed, and with the MQTT core several boards will work together very easily.

If the ESP board cost $30 I think a shield would be a great idea, but I can pick up ESP boards for $2-$5, making it easy enough to just have 3 boards running different devices.

posicat, 2018-01-25:
What are people's priorities for a PCB, what modules do you want to have access, and which ESP board are you going to use?  #1 below I'm building for sure, #2 is still open for input on what people need.

1) ESP-01
- Based on an ESP-01 board
- 5V power, 3.3V regulator
- IR transceiver
- I2C (a board I need at home, but others might want as well.)

2) ESP-12F : (current design, what are your preferences?)
- Based on an ESP-12F board
- 5V power, 3.3V regulator
- IR transmit/receive
- I2C 3.3v and 5v with level shiftier
- 433Mhz transmit/receive
- Pins remaining : 8 IO and ADC, Here are the possibilities...
  - Programming connector header
  - Bluetooth [2 pins]
  - SPI Breakout connection(s) (RFM69 Module, or other possible SPI devices) [1(or more) pins + MOSI/MISO]
  - DHT11/22 [1 pin]
  - WS2812 RGB LED  I can add pads to solder 3 of these in for indicators, and a socket to expand to an entire string.) [1 pin]
  - Generic analog result from the ADC (would need code as well) [1 pin]
  - All remaining pins brought out to a header for future expansion.

posicat, 2018-01-28:
I received my PCBs yesterday and got around to soldering them this evening, I can confirm the circuit design on the ESP-01 board works properly : https://easyeda.com/posicat/OMG_ESP_01_PCB-445e573315b543108581500a4d4974be 

*** Warning *** I used VERY small surface mount components to keep the design small, you must have a steady hand and a good soldering tool to build these boards.  

You can order parts and PCB's for that board by scrolling down to the image of the PCB, and opening it in the editor.  From there click the Icon with the G on the top to order the boards, and the BOM button to order the components.

I may have some kits to offer in the future after I build a few more for myself.  If you just one one to beta-test drop me an email posicat@pobox.com.  If you want more than one, you're probably better off just ordering some via the link above.

1technophile, 2018-01-28:
Nice to see that there is improvment on the hardware!
In my side I'm more interested in ESP12F with at least (if you have enough pins...)

- 5V power, 3.3V regulator
- IR transmit/receive
- 433Mhz transmit/receive (superheterodyne)
- WS2812 RGB LED
- RFM69HCW
- programming connector

posicat, 2018-01-28:
I'll post this on the Wiki and some pictures so folks can find it.

posicat, 2018-02-26:
Just got my tax refund, picked up a few ESP8266 boards, need to some 433mhz radios (any one considered the best model?) so I can try and prototype a board for the ESP-12F.

There were some problems with the pinouts for the D3 mentioned earlier, have those been resolved?

jumpalottahigh, 2018-02-27:
I have some of those (superheterodyne):

https://www.aliexpress.com/item/433MHz-100-Meters-Wireless-Module-Kit-ASK-Transmitter-STX882-ASK-Receiver-SRX882-2Pcs-Copper-Spring-Antenna/32637181317.html?spm=a2g0s.9042311.0.0.JK1jKr

and I also have some of those cheaper ones (regular):

https://www.aliexpress.com/item/1Lot-1-pair-2pcs-Best-prices-433Mhz-RF-transmitter-and-receiver-link-kit-for-Arduino/32318951712.html?spm=a2g0s.9042311.0.0.JK1jKr

1technophile, 2018-02-27:
D3 is not a good idea for RF receiver, we should switch with another pin. D2 could be a option if you are not planning to use I2C bus
Global pinout is summarized here:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit?usp=sharing

posicat, 2018-03-12:
Working on a board for this, and I'm running out of pins.  I have the I2C setup as the programming lines as well so they're dual-purpose since it looks like the RX/TX lines can be used just fine as I2C.  5V USB power, 3.3V regulator, RFM69HCW hooked up, IR Transmit/Receive and 433Mhz  Transmit/Receive.  No leftover pins for the WS2812 status LED unfortunately, but the ADC pin is free still.

Several options are available

Option A
The WS2812 status LED could be moved over to the I2C bus, I already have a controller board from another project designed especially for running WS2812's off of I2C using an ATTiny85, this would also give us 2 more pins for interfacing.  I could even write code that would return data via I2C in JSON format if necessary.  An ATTiny84 daughter board could also be built that would offer even more IO pins if we wanted to add temperature, humidity, light, etc.

Option B
Design a second PCB with it's own ESP-12F that does whatever doesn't fit on this board.  ESP's are cheap, this is probably the best ideal.  At that point we could sacrifice one of the functions of the current board (Personally I'd drop both of the IR lines so we could add the WS2812 AND a DHT22 temperature sensor since we already have a PCB for the IR.

Option C
I could also design the board with optional sections "jumper A+B for IR, C+D for 433mhz, etc."  That would make the PCB itself the most flexible, as people could add only what they needed.  It would be flexible, but messy.

Option D

Pull off the IR completely (we have a board for that) and break out the remaining free pins to a WS2812  and a header for whatever people want to hook up and call it the Radio Board.  Add BLE.
    Radio Board  : WS2812, I2C, BLE, RFM69HCW and 433Mhz
    IR Board (Built already) : WS2812, I2C and IR Transmit/Receive
    Sensor Board : WS2812, Light, Temp, Humidity, Analog Pin, Whatever.

To me Option D is the way to go, the PCB's are fairly inexpensive to product, and it's unlikely the RF boards (Centralized in the house for best signal) will be used in the same places that the IR boards are (Line of sight to components) or the sensors (in different rooms for whole house monitoring).

broekema, 2018-04-13:
I've been playing with easyeda to try and move my main hub from breadboard to PCB. Required functionality:

1. IR Send only (4x LED, driven by BC547C transistors)
2. 433 Send only
3. I2C BME280 
It would be nice to also mount a TSL2561 ambient light sensor module, so I've added that as well. I'm basing this PCB on a Wemos D1 mini module.

A work of progress is available here:
https://easyeda.com/broekema/MQTT-hub
Note that the resistors are not the right value yet.

posicat, 2018-04-13:
Nice board!  If you want to free up a few pins, and get greater LED
control, you might look into the WS2812B LEDs, I know one of the other
people on here was already looking at including that library for those.

1technophile, 2023-06-12:
Closing this as PCB is not in the scope of this repository