1technophile/OpenMQTTGateway #66: BLE Problem

aferraz, 2017-06-22:
Hi all,
 I have a little problem.
I'm using a NodeMCU and when connect the HM 10 the board don;t boot with the following error:
 ets Jan  8 2013,rst cause:2, boot mode:(7,6)

waiting for host

If I disconnect the HM 10 the board boot again.

Any suggestion ?

Thanks
 Alessandro

1technophile, 2017-06-23:
Hi @aferraz,

Ciuld yiu give the following details:
-is the hm10 a keyes one
-what is your nodemcu and esp version/model
-is the nodemcu rebooting when you connect the hm10 only when openmqttgateway is loaded or do you have the same behaviour with other programs?
-is the hm10 connected to the pins given in the wiki?
-did you checked the wiring tx/rx

aferraz, 2017-06-23:
Hi,
 - it's keyes 
- nodemcu 1, esp8266 12-E
- if connected after boot all work fine
- I follow the wiki schema for pin connection
- tx/rx is ok, I notice that the hm10 tx pin is the problem, if disconnected before boot the mcu boot, reconnecting after correct boot the hm10 work 

Thanks

1technophile, 2017-06-23:
Hello,

Could you try with I2C_Wiring in user_config.h ?
Uncomment l36
`#define I2C_Wiring`
Comment l37 
`//#define Classic_Wiring`

And change the physical wiring corresponding to the end of user_config.h file

aferraz, 2017-06-23:
Hello @1technophile,
 same error with HM10 connected

Regards
 Alessandro

1technophile, 2017-06-23:
In Arduino IDE --> board manager could point me out which version of ESP8266 you are using.

Could you check also your HM10 firmware version

aferraz, 2017-06-24:
Hi,
 on Arduino the ESP8266 version is 2.3.0
I don't know how to check HM10 firmware, can you help me ?

Thanks

aferraz, 2017-06-24:
@1technophile found it, firmware version HMSoft V540

1technophile, 2017-06-24:
ok just for others, to check your hm10 firmware version upload a serial sketch to the nodemcu (this will enable to talk directly to the hm10) and launch the command:
`AT+VERR?`

Mine is returning:
`HMSoft V545`

Could you try to update to V545?

1technophile, 2017-06-24:
and with v549?

aferraz, 2017-06-24:
I can't flash with 5.49, after flashing the module stay in flashing mode

aferraz, 2017-06-25:
@1technophile I don't know if is a problem but on the same MCU I have also RF and IR

1technophile, 2017-06-25:
Could you try with another sketch , not openmqttgateway and check if you uve the sme issue

aferraz, 2017-06-26:
I try this http://www.instructables.com/id/ESP8266-HM-10-IBeacon-Proximity/

Same problem, at this point could be my HM10

1technophile, 2017-06-26:
I have a second HM10 i will test with it

1technophile, 2017-06-26:
Tested with a second one hm10 with firmware v545 and it works. Do you have a second nodeMCU to test with?

aferraz, 2017-06-27:
Hi @1technophile with another nodemcu it work. So is the MCU the problem.
Thank you

closed 2017-06-27 by aferraz

1technophile, 2017-06-27:
Glad to hear that