1technophile/OpenMQTTGateway #67: IR Transmitter

aferraz, 2017-06-28:
Hi all,
 someone can tell me if this is usable on the gateway ?
http://www.gearbest.com/transmitters-receivers-module/pp_346872.html

Thanks

HannesDi, 2017-06-28:
1st. i think its not possible to use this, because the gateway pulse the signal to the ir led 
and this looks like it need power an a dedicated signal pin.

so its probally possible to bridge the ground and signal pin for make this functional but not 100% sure.

Better Question is.. Why use this? 

A IR Diode do the same job but cheaper...

and you also need the resitor and the transistor so nothing won from my perspective...

Best Regards
Hannes

aferraz, 2017-06-28:
Thanks @HannesDi ,
 I wanted to use this because I'm a bit denied with the soldering tool ;-)

Regards
 Alessandro

closed 2017-06-28 by aferraz