1technophile/OpenMQTTGateway #69: HN3031 RF Switch

donnib, 2017-06-30:
Hi,
I have one of these switches and id like to buy more if i can get them to work with this. Unfortunately i don't have an remote for it so i can't replicate the signal to learn it. Is there another way to do this ? 
I read that in the first 30sec of it turned on it's in learning mode so i should be able to send a code to it that it learns but i need to know the protocol and what to send, any ideas ?

The product : Sartano HN3031

1technophile, 2017-06-30:
Hi,

I don't think it will work with RCswitch (the library used by the gateway), and even if it work it can be tricky to find the code. My advise is to go to devices compatible:
https://community.home-assistant.io/t/433tomqttto433-gateway-device-list/7819

donnib, 2017-07-01:
So i did find a remote which i programmed to control it. I tried to look at home/433toMQTTAdvanced but nothing comes in. I can see codes come in on home/433toMQTT but kinda weird same button will some times give two different codes. I see that home/433toMQTT/protocol gives 2 and bits 32 and length 714 or 716 and i tried so send the codes i got but the switch does not react. Any ideas what could be wrong ?

1technophile, 2017-07-01:
Did you try by specifying protocol number, bits length and pulselength or only the RF code?

donnib, 2017-07-01:
I tried specifying all of them using the home/commands/PLSL_714/433_2/RFBITS_32

1technophile, 2017-07-11:
I advise to go at rcswitch library level in your case by trying the examples given with the library and if it doesn't work submit an issue on their github

donnib, 2017-07-16:
@1technophile i can see you are using these https://www.leroymerlin.fr/v3/p/produits/lot-de-4-prises-telecommandables-interieure-idk-e1400980485 which looks exactly like mine. Can you reveal what you send to those ? like length, protocol and bits and eventually the code (if you don't want to reveal that then maybe just an example in the same number length) so i can try it out and see if it works on mine.

1technophile, 2017-07-16:
@donnib hello, there is some dip switched to program on these plugs, could you send me a picture of one of them, I can deduct after the corresponding code

DenDeze, 2017-07-17:
I had the same, sometimes returning 2 codes. After sending the codes i could filter out the wrong one. Did you tried sending the codes?

1technophile, 2017-07-17:
I mean if you send me a picture of your dip switch position on the plug I can deduct directly your codes by positionning mine with the same position.
I advise to use the tchat room for this kind of issue, it would be more appropriate

donnib, 2017-07-17:
@DenDeze i believe i did, i'll have to try again to see if i did something wrong.

DenDeze, 2017-07-17:
Can you tell if you used the correct mqtt topic?

When you subscribe to # you subscribe to all the topics on the server. Easy for debugging.

Verstuurd vanaf mijn iPhone

ENGIE Mail Disclaimer: http://www.engie.com/disclaimer/disclaimer-nl.html

1technophile, 2017-07-19:
@donnib maybe you can close this issue as you succeeding in controling the switch with openmqttgateway

donnib, 2017-07-24:
Yes done.

closed 2017-07-24 by donnib