1technophile/OpenMQTTGateway #70: how do i test the sender ?

donnib, 2017-07-03:
I mean if i don't have a sender and send something to home/commands/MQTTto433 then i see something on home/433toMQTT so obviously i see something that has nothing to do with the real sender so how do i check whether it works or not ? Is there a way ?

1technophile, 2017-07-03:
what you are seeing in home/433toMQTT is the acknowledgement that the gateway received your command, it enables home assistant to update the state switch.
I this is anoying you you can change in user_config.h the value of 
`#define subjectGTWRFtoMQTT "home/433toMQTT"`

to another subject

To be sure that the gateway is sending verify the following things:
-check your emitter power voltage with a multimeter
-check your wiring

If these two points are correct, upload the receive example of rcswitch library to an arduino or an esp8266 with a receiver and listen while you send command to the gateway.

May I advise you the chat room for support:
https://gitter.im/OpenMQTTGateway/Questions_support

donnib, 2017-07-04:
@1technophile thank you, i'll dig up another arduino or just make another gateway and make the test. I'll keep mind the chatroom for further questions.

closed 2017-07-04 by donnib