1technophile/OpenMQTTGateway #71: How to identify BT LE device

aferraz, 2017-07-11:
Hi,
 how can I identify a device from the bluetooth address ?
I have iphone and ipad but the mac in the mqtt messange and the mac on the device settings does not match. 

Thanks
 Alessandro

1technophile, 2017-07-11:
Hello,

The gateway handle beacons detection, are you sure your mac and iphone diffuse a mac adress as beacons without any applications installed for that ?

aferraz, 2017-07-11:
Hello,
 I’m not sure but I can see different mac’s in my home and I live outside the city.

Thanks

1technophile, 2017-07-11:
You could try to switch on your iphone or mac and see what's going on

aferraz, 2017-07-12:
Hi,
 I make some tests.
iPhone, and iPad also, announces itself to the hm10, also without a specific app installed.
The strange thing is that the MAC address is always different.

1technophile, 2017-07-12:
Hi,

Could you upload this piece of code to your esp8266 with the hm10 and see what the serial monitor is saying:

```
    This code is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This code is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.

#include <SoftwareSerial.h>

#define STRING_MSG "OK+DISC"
#define QUESTION_MSG "AT+DISI?"
#define RESPONSE_MSG "OK+DISIS"
#define SETUP_RES_MSG "RESPONSE_MSG"

SoftwareSerial softserial(D7, D6);
//SoftwareSerial softserial(10, 11);
unsigned long time1 = 0;

void setup() {
  Serial.begin(115200);
  softserial.begin(9600);
  softserial.print(F("AT+ROLE1"));
  delay(100);
  softserial.print(F("AT+IMME1"));
  delay(100);
  softserial.print(F("AT+RESET"));
  delay(100);
}

void loop() {
  while (softserial.available() > 0) {
    #ifdef ESP8266
      yield();
    #endif

    String response = softserial.readString();
    Serial.println(response);
    if (response.indexOf(STRING_MSG)>=0)
    {
      response.replace(RESPONSE_MSG,"");
      int device_number = response.length()/78;
      Serial.println(device_number);
      
      for (int i=0;i<device_number;i++){
           String onedevice = response.substring(0,78);
           onedevice.replace(STRING_MSG,"");
           Serial.println(onedevice);
           String company = onedevice.substring(1,9);
           Serial.println(company);
           String uuid = onedevice.substring(10,42);
           Serial.println(uuid);
           String others = onedevice.substring(43,53);
           Serial.println(others);
           String mac = onedevice.substring(54,66);
           Serial.println(mac);
           String rssi = onedevice.substring(67,71);
           Serial.println(rssi);
           response = response.substring(78);
        }
      }
    if (response.indexOf("OK+RESET")>=0)
    {
      Serial.println("connection OK");
  }
  }
  if (millis() > (time1 + 10000)) {//retriving value of adresses and rssi
       time1 = millis();
       softserial.print(F(QUESTION_MSG));
  }
    /*while (Serial.available() > 0) {
    softserial.write(Serial.read());*/
  }
}
```

aferraz, 2017-07-12:
Hi,
 this is the output:
OK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:607626CA1163:-057
1
:00000000:00000000000000000000000000000000:0000000000:607626CA1163:-057
00000000
00000000000000000000000000000000
0000000000
607626CA1163
-057
OK+DISCE
0
OK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:607626CA1163:-040
1
:00000000:00000000000000000000000000000000:0000000000:607626CA1163:-040
00000000
00000000000000000000000000000000
0000000000
607626CA1163
-040
OK+DISCE
0
OK+DISIS
OK+DISCE
0
OK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:607626CA1163:-048
1
:00000000:00000000000000000000000000000000:0000000000:607626CA1163:-048
00000000
00000000000000000000000000000000
0000000000
607626CA1163
-048
OK+DISCE
0

Regards

1technophile, 2017-07-12:
everything seems fine, are you sure that you compare it with BLE mac adress and not WIFI mac adress on the devices?

aferraz, 2017-07-12:
I compare it with bluetooth and I don't know if is the same for beacon but I imagine yes.
The strange thing is that if I dosable bluetooth in iPhone and re enable it the Mac address is different

1technophile, 2017-07-12:
what is your ios version?

aferraz, 2017-07-12:
ios 11 on iphone, ios 10.3 on ipad

1technophile, 2017-07-12:
I think they are using randomization of their BLE mac adress that's why you never see the same adress and also explain why the adress is not the same as your real adress.
This is to avoid tracking of your devices.

aferraz, 2017-07-12:
I think the same thing.

DenDeze, 2017-07-14:
Because of security apple does not provide you with the dame UUID over time.

You have tor nu a virtual beacon like this: https://community.estimote.com/hc/en-us/articles/200908836-How-to-turn-my-iPhone-into-a-Virtual-Beacon-

Of write tour OWN app where you can set a fixen UUID.

aferraz, 2017-07-14:
This is a problem :-(

1technophile, 2017-07-14:
yes there is several apps available in android and ios store to simulate beacons.
In my side I'm not planning to use my phone for presence detection but a nut attached to my keys and some beacons placed on my vehicles

1technophile, 2017-07-19:
@aferraz did you try installing a beacon simulator?

aferraz, 2017-07-19:
@1technophile  yes but with no luck. I think iOS is the problem but I don't have an Android device for make some tests.

1technophile, 2017-07-19:
With android it's working, I tested it with this app:
https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=fr

aferraz, 2017-07-19:
@1technophile perfetc. At this poin I close this issue. It's clear that this is not a software problem.

1technophile, 2017-07-19:
wiki updated

1technophile, 2017-08-20:
from HA forum:
https://community.home-assistant.io/t/433mhz-infrared-ir-to-and-from-mqtt-on-esp8266/6779/237?u=1technophile

> According to the hm10 data sheet if you use AT+SHOW1 you should get the device name as well which will be handy as iphone keeps changing mac, I have tried it but doesn’t seems to get it working, have any of you tried it before?

aferraz, 2017-08-23:
Hi @1technophile,
 I try with command AT+SHOW1 but no response from device, I would expect at least OK.
After reconnecting the module to MCU I have the same behavior, a random MAC Address.

Regards
 Alessandro

1technophile, 2017-08-23:
Hi @aferraz ,

THanks for the feedback, could you give me the version of hm-10 firmware your are using?

aferraz, 2017-08-23:
HMSoft V545

1technophile, 2017-09-12:
I reclose the issue as even we are not getting the name of apple devices

closed 2017-09-12 by 1technophile