1technophile/OpenMQTTGateway #75: Byron

scargill, 2017-07-20:
Is there a way to add devices to this - got the ESP8266 version up and running - handles a little 4-button remote control 100% reliably so it is working..

But UTTERLY ignores a ByronSX35 433Mhz doorbell.   I know that the RF433 kit handles this as I have one so it's not as if it is some completely alien format.

Thoughts?

1technophile, 2017-07-22:
Hello, by rf433 kit you mean rflink library ?

scargill, 2017-07-22:
I’m sorry – yes I meant the RFLink Library…

1technophile, 2017-07-23:
the options are the following:
-integrate the Byron decoding at rcswitch level (the library used by the gateway to decode & generate RF), for that I suggest to request for an enhancement into [rcswitch repo](https://github.com/sui77/rc-switch). 
-integrate into openmqttgateway an open source  library that decode already byron protocol and reproduce it, we didn't find it yet. Nevertheless ther eis maybe something to explore based on Nigel comment on this page:
http://arduinobasics.blogspot.fr/2014/06/433-mhz-rf-module-with-arduino-tutorial.html
I asked him how he did it.
-build a library from scratch ...

I think this request is a duplicate of #26

1technophile, 2019-01-31:
We've just integrated the Pilight library, adding by the way more compatible [protocols](https://manual.pilight.org/protocols/433.92/index.html), if you are still interested you can test the last version of OMG[ v0.9](https://github.com/1technophile/OpenMQTTGateway/releases/tag/v0.9)

1technophile, 2019-04-15:
I'm closing this issue, if you have feedback to give with the use of ZgatewayPilight don't hesitate to add it here

closed 2019-04-15 by 1technophile