1technophile/OpenMQTTGateway #76: Advanced 433 SIgnal wrongly triggers Topic

AKA-47, 2017-07-27:
Hi.
i have a problem with some advanced states which wrongly trigger my openhab.
Example:

My devices have code 15 for ON and 14 for OFF.

home/433toMQTT/protocol 1
home/433toMQTT/bits 24
home/433toMQTT/length 205
home/433toMQTT 15

home/433toMQTT/protocol 1
home/433toMQTT/bits 24
home/433toMQTT/length 205
home/433toMQTT 14

My openhab Item looks like this: 

Switch MQTT_Switch_NOPROG1 "Lampe J1" {mqtt=">[broker:home/commands/MQTTto433:command:ON:15], >[broker:home/commands/MQTTto433:command:OFF:14], <[broker: home/433toMQTT:state:ON:15],

if i now receive a signal like this:

home/433toMQTT/protocol 13
home/433toMQTT/bits 6
home/433toMQTT/length 407
home/433toMQTT 14

this triggers my switch in openhab - bit this is not my wifi plug rf signal which is:
home/433toMQTT/protocol 1
home/433toMQTT/bits 24
home/433toMQTT/length 205

so i tried to change the topics to something like 433toMQTTAdvanced:

home/433toMQTTAdvanced/protocol 13
home/433toMQTTAdvanced/bits 6
home/433toMQTTAdvanced/length 407
home/433toMQTT 14

...without success...

maybe it makes sense to publish more specific like this? Or are there other ideas what i can do to to be sure it is the correct signal?

home/commands/MQTTto433/protocol1/bits24/length205/15
or 
home/commands/MQTTto433/protocol1/bits24/length205/14

AKA-47, 2017-07-27:
hmm... it worked for me like this - not nice but specific enough for me

#define subjectRFtoMQTTADVINFO "home/433toMQTTAdvancedINFO"

        String adv = String(MQTTprotocol) +  "-" + String(MQTTbits) + "-" + String(MQTTvalue);
        client.publish(subjectRFtoMQTTADVINFO,(char *)adv.c_str());  

Now i get a publish like this: 
home/433toMQTTAdvanced/protocol 1
home/433toMQTTAdvanced/bits 24
home/433toMQTTAdvanced/length 205
home/433toMQTTAdvancedINFO 1-24-15
home/433toMQTT 15

home/433toMQTTAdvanced/protocol 1
home/433toMQTTAdvanced/bits 24
home/433toMQTTAdvanced/length 205
home/433toMQTTAdvancedINFO 1-24-14
home/433toMQTT 14

Switch MQTT_Switch_NOPROG1 "Lampe J1" {mqtt=">[broker:home/commands/MQTTto433:command:ON:15], >[broker:home/commands/MQTTto433:command:OFF:14], <[broker: home/433toMQTTAdvancedINFO:state:ON:1-24-15], 
<[broker: home/433toMQTTAdvancedINFO:state:OFF:1-24-14]", autoupdate="false"}

AKA-47, 2017-07-27:
had to do some more modifcations to get the backwards channel advertisment running...

```
```
#ifdef ZgatewayRF

#include <RCSwitch.h> // library for controling Radio frequency switch

RCSwitch mySwitch = RCSwitch();

void setupRF(){

  //RF init parameters
  mySwitch.enableTransmit(RF_EMITTER_PIN);
  mySwitch.setRepeatTransmit(RF_EMITTER_REPEAT); 
  mySwitch.enableReceive(RF_RECEIVER_PIN); 
  
}

boolean RFtoMQTT(){

  if (mySwitch.available()){
    trc(F("Receiving RF"));
    unsigned long MQTTvalue = 0;
    String MQTTprotocol;
    String MQTTbits;
    String specificSignal;
    String MQTTlength;
    MQTTvalue = mySwitch.getReceivedValue();
    MQTTprotocol = String(mySwitch.getReceivedProtocol());
    MQTTbits = String(mySwitch.getReceivedBitlength());
    MQTTlength = String(mySwitch.getReceivedDelay());
    mySwitch.resetAvailable();
    if (!isAduplicate(MQTTvalue) && MQTTvalue!=0) {// conditions to avoid duplications of RF -->MQTT
        trc(F("Adv data RFtoMQTT"));
        client.publish(subjectRFtoMQTTprotocol,(char *)MQTTprotocol.c_str());
        client.publish(subjectRFtoMQTTbits,(char *)MQTTbits.c_str());    
        client.publish(subjectRFtoMQTTlength,(char *)MQTTlength.c_str());   
        String adv = String(MQTTprotocol) +  "-" + String(MQTTbits) + "-" + String(MQTTvalue);
        client.publish(subjectRFtoMQTTADVINFO,(char *)adv.c_str());  
        trc(F("Sending RFtoMQTT"));
        String value = String(MQTTvalue);
        trc(value);
        boolean result = client.publish(subjectRFtoMQTT,(char *)value.c_str());
        
        if (repeatRFwMQTT){
            trc(F("Publishing RF for repeat"));
            client.publish(subjectMQTTtoRF,(char *)value.c_str());
        }
        return result;
    } 
  }
  return false;
}

void MQTTtoRF(char * topicOri, char * datacallback) {

  unsigned long data = strtoul(datacallback, NULL, 10); // we will not be able to pass values > 4294967295

  // RF DATA ANALYSIS
  //We look into the subject to see if a special RF protocol is defined 
  String topic = topicOri;
  int valuePRT = 0;
  int valuePLSL  = 0;
  int valueBITS  = 0;
  int pos = topic.lastIndexOf(RFprotocolKey);       
  if (pos != -1){
    pos = pos + +strlen(RFprotocolKey);
    valuePRT = (topic.substring(pos,pos + 1)).toInt();
    trc(F("RF Protocol:"));
    trc(String(valuePRT));
  }
  //We look into the subject to see if a special RF pulselength is defined 
  int pos2 = topic.lastIndexOf(RFpulselengthKey);
  if (pos2 != -1) {
    pos2 = pos2 + strlen(RFpulselengthKey);
    valuePLSL = (topic.substring(pos2,pos2 + 3)).toInt();
    trc(F("RF Pulse Length:"));
    trc(String(valuePLSL));
  }
  int pos3 = topic.lastIndexOf(RFbitsKey);       
  if (pos3 != -1){
    pos3 = pos3 + strlen(RFbitsKey);
    valueBITS = (topic.substring(pos3,pos3 + 2)).toInt();
    trc(F("Bits nb:"));
    trc(String(valueBITS));
  }
  
  if ((topic == subjectMQTTtoRF) && (valuePRT == 0) && (valuePLSL  == 0) && (valueBITS == 0)){
    trc(F("MQTTtoRF default"));
    mySwitch.setProtocol(1,350);
    mySwitch.send(data, 24);
    // Acknowledgement to the GTWRF topic
    boolean result = client.publish(subjectGTWRFtoMQTT, datacallback);

    String adv = String(1) +  "-" + String(24) + "-" + String(data);
    client.publish(subjectRFtoMQTTADVINFO,(char *)adv.c_str());  

        
    if (result)trc(F("Ack published"));
    
  } else if ((valuePRT != 0) || (valuePLSL  != 0)|| (valueBITS  != 0)){
    trc(F("MQTTtoRF user parameters"));
    if (valuePRT == 0) valuePRT = 1;
    if (valuePLSL == 0) valuePLSL = 350;
    if (valueBITS == 0) valueBITS = 24;
    trc(String(valuePRT));
    trc(String(valuePLSL));
    trc(String(valueBITS));
    mySwitch.setProtocol(valuePRT,valuePLSL);
    mySwitch.send(data, valueBITS);
    // Acknowledgement to the GTWRF topic 
    boolean result = client.publish(subjectGTWRFtoMQTT, datacallback);// we acknowledge the sending by publishing the value to an acknowledgement topic, for the moment even if it is a signal repetition we acknowledge also
    
    String adv = String(valuePRT) +  "-" + String(valueBITS) + "-" + String(data);
    client.publish(subjectRFtoMQTTADVINFO,(char *)adv.c_str());  
    
    if (result){
      trc(F("MQTTtoRF OK ack published"));
      trc(String(data));
      };
  }
  
}
#endif 
```
```

1technophile, 2017-07-27:
Nice one, i never had this use case. Thanks for pointing this. This will be corrected when i will implement json format payload. I will keep this issue open until that.

1technophile, 2017-07-28:
#49

AKA-47, 2018-01-31:
any news about the json output?

1technophile, 2018-02-01:
it's the next point on my TODO

AKA-47, 2018-03-18:
..thanks for this... Will try this weekend

1technophile, 2018-11-10:
you can now receive RF with json formatting (see the current master code)

closed 2018-11-10 by 1technophile