1technophile/OpenMQTTGateway #77: Bt is not working on esp8266

killermatz, 2017-07-27:
Pin in User_config.h is not correct.

Set it to 12 and 13.

Serial Monitor say "connection hm-10ok" but dont detect any bluetooth device.

1technophile, 2017-07-27:
Hello,
 are you using keyes hm10 ? 
You get serial connection ok with which pin ? 
What are your hm10 firmware version ?

killermatz, 2017-07-27:
Im using hm10 soft version 549
Serial monitor in arduino ide say connected

Set RX to gpio 12
Set TX to gpio 13

MfG Martin

1technophile, 2017-07-28:
hm10 detect bluetooth low energy beacons and not bluetooth devices, did  you tested with beacons or a beacon simulator?

killermatz, 2017-07-28:
Did you mean that the hm10 does not detect smartphones???

No i didnt test it with beaconsimulator.

MfG Martin

1technophile, 2017-07-28:
it detect smartphone if your smartphone is compatible with BLE beacons simulation and install a beacon simulator app

1technophile, 2017-08-01:
Did you advance in your test?

1technophile, 2017-08-05:
I'm closing this issue as I have no news, please reopen it if necessary

closed 2017-08-05 by 1technophile