1technophile/OpenMQTTGateway #80: Make openmqttgateway compatible with happybubbles presence server

1technophile, 2017-07-31:


yzg011, 2017-09-19:
it is good idear，i also need this

jxjhheric, 2017-10-19:
Hi ,any plan to get it done?

1technophile, 2017-10-19:
If you want to make it in top of the list please vote for it with the thumbs up on the issue description

jxjhheric, 2017-12-05:
Do This Help?https://github.com/RiRomain/Happy_Bubble_ESP32_Node/

PetricaM, 2018-01-05:
Hi Florian,

With a bunch of BLE gateways I managed to play along with Happy Bubbles server.

From what I could make of https://www.happybubbles.tech/presence/docs/mqtt/ there are three types of messages that can be processed by the server.

For iBeacon messages (which seemed the easier :) ), the server looks in topic happy-bubbles/ble/<the-hostname>/ibeacon/<the-iBeacon-UUID> and the payload would be like this:

{
    "hostname": "living-room",
    "beacon_type": "ibeacon",
    "mac": "dd6ed85b7a80",
    "rssi": -92,
    "uuid": "e2c56db5dffb48d2b060d0f5a71096e0",
    "major": "0068",
    "minor": "0068",
    "tx_power": "c5"
}

Based on this, the OpenMQTT BLE payloads can be easily resent through an automation to the topic that Happy Bubble can use:

```
- alias: testble
  initial_state: True
  trigger:
    - platform: mqtt
      topic: 'home/+/BTtoMQTT/+' 
  action:
    - service: mqtt.publish
      data_template:
        topic: "happy-bubbles/ble/{{ trigger.topic.split('/')[-3] }}/ibeacon/{{trigger.topic.split('/')[3]}}"
        payload_template: >
          {
            "hostname":"{{trigger.topic.split('/')[-3]}}",
            "beacon_type":"ibeacon",
            "mac":"{{ trigger.topic.split('/')[3]}}",
            "rssi":{{trigger.payload}},
            "uuid":"{{ trigger.topic.split('/')[3]}}",
            "major":"0068",
            "minor":"0068",
            "tx_power": "c5"
          } 

```
I've used as hostname the name of the gateway from #define Gateway_Name and then the topic is #define subjectBTtoMQTT "home/" Gateway_Name "/BTtoMQTT/" but it can be easily modified by defining also a Room_Name in addition to Gateway_Name and use that as topic for BT.

As I didn't find UUID to be sent by the gateway to MQTT, I used BLE MAC as UUID and it seems that there is no check for 32 char UUID length. 

The problem is almost solved as major and minor are only identifiers (case the UUID is the same for multiple BLE devices, however as I've used MAC as UUID it should be unique). 

The sole issue that I think remains is the tx_power for which I didn't find any reference in the sketch.  Is it, by any chance, included in the signal received by the gateway so that it could be published to  MQTT as it would improve reliability of the signal processing?

1technophile, 2018-01-07:
Hello @PetricaM ,

Thanks for your investigations, the ESP32 BLE library enable to retrieve TX power, I added the function to the master (and also manufacturerdata retrieving).
Nevertheless with my NUT and mi flora the BLE library doesn't publish TX power... don't know why
You can test my last commit if you want.

With HM10 I didn't found a way to get tx power.

1technophile, 2018-11-28:
I don't think this is still needed as we have room presence integration in HA, if you still need it don't hesitate to reopen it.

closed 2018-11-28 by 1technophile