1technophile/OpenMQTTGateway #81: Sonoff rf bridge

prahjister, 2017-08-02:
Any plans to port over to new sonoff rf bridge?

1technophile, 2017-08-05:
need to buy the device first, good idea

prahjister, 2017-08-07:
Email me prahjister0 at the gmail domain and I will see if I can get one sent. I would send the one that I have but it looks like you are in France. I would try to port but it is beyond my skill set. Hardware seems solid on the review sample I have. Do you have the ability to reverse engineer. Looks like 2 chips are working together to make this work?

1technophile, 2017-08-07:
if it is based on esp8266 like the other sonoff it shouldn't be a problem.
I sent you an email.

prahjister, 2017-08-07:
I never got the email. prahjister0

prahjister, 2017-09-20:
Did they ship it and you receive it?

1technophile, 2017-09-20:
Not received it yet

1technophile, 2017-10-04:
Received  !

prahjister, 2017-10-04:
Awesome. Maybe you can improve latency over other firmware that I'm using.
Also still getting duplicates.

prahjister, 2017-10-13:
How goes it?

1technophile, 2017-10-14:
I didn't put my hands on it still

1technophile, 2017-11-16:
Begining to look at sonoff rf,
there is already some interesting work about it especialy:
http://tinkerman.cat/hacking-sonoff-rf-bridge-433/

At this time I'm able to receive RF with a mixed code between tinkerman one and mine (mainly tinkerman ;-)).

prahjister, 2017-11-16:
Great. I'm using https://github.com/arendst/Sonoff-Tasmota

Latency and still getting duplicates. I think the way he is stopping duplicates is fooled when it missintepretes one of the duplicate transmission. I haven't looked into it much

1technophile, 2017-11-17:
You should take a look at @xoseperez code he is well removing the duplicates code (key RF_RECEIVE_DELAY)
http://tinkerman.cat/hacking-sonoff-rf-bridge-433/

1technophile, 2017-11-24:
done based on Xose Pérez work ESPURNA

1technophile, 2017-11-24:
Sending RF
`sudo mosquitto_pub -t home/commands/MQTTtoSRFB -m 1314884`

Receiving RF
```
home/433toMQTT 34009
home/SRFBtoMQTT/Raw 2A80015E042E0084D9
home/SRFBtoMQTT 34009
home/SRFBtoMQTT/Tsyn 10880
home/SRFBtoMQTT/Thigh 350
home/SRFBtoMQTT/Tlow 1070
```

closed 2017-11-24 by 1technophile

1technophile, 2017-12-23:
@prahjister Hello, finally did you get it working with OpenMQTTGateway?

MACscr, 2018-01-05:
I actually have espurna installed, but i dont think it can work as a receiver that way. Seems like it couldnt do both receiver and send. Only send after its learned. Im looking for something that can do both. At least receive since my broadlink can already do the sending part. I need the receive so that I can actually use RF remotes with it. Also, the RF Bridge is esp8265, not 66 afaik.

1technophile, 2018-01-05:
OpenMQTTGateway will do the job you are requesting on the platform you want (sonoff rf bridge, ESP8266, arduino, ESP32)

xoseperez, 2018-01-28:
@MACscr You can send arbitrary codes using the 0xA5 action, no need to have them learned beforehand. Using ESPurna you can send the code to "<base_topic>/rfout/set" topic and the device will broadcast it.

MACscr, 2018-01-29:
@xoseperez im not sure what you are talking about. i need the bridge to act as a RF/MQTT gateway. So it can receive commands from my RF remote and pass that to my MQTT gateway. As @1technophile and I have figured out, it doesnt reliable work that way. It works fine for learning a code and sending that code, but it cant consistently receive commands. seems the same remote (etekcity) buttons are inconsistent on which code they send.

AKA-47, 2018-01-31:
i saw the same problem like @MACscr on some cheap self learning plugs. if i debug this with my gateway based on STX88 receiver it shows that the remote sends 5 different codes of which my plug only reacts on one of this if i resend it via my STX Gateway. Receiving or Sending the same code via RFbridge doesnt work. it only works on very static transmissions as i see so we maybe need to maintain an additional section in the compatibility list for this "additional microprocessor" based device