1technophile/OpenMQTTGateway #83: Configuration Webserver

Trickx, 2017-08-07:
This is a feature request for a webserver to configure the device.
Currently the sources need to be touched for every individual device.
What about a webinterface to store the configuration into EEPROM?

- If WLAN settings are not properly read from EEPROM or invalid change into AP mode.
- Provide IP addresses to connected clients (DHCP) server.
- Announce webservie via mDNS (bonjour).
- Provide webinterface to configure MQTT and WiFi settings.
- Provide webinterface to configure services.

cosmocracy, 2017-11-02:
Definitely a use case for integrating [WiFiManager](https://github.com/tzapu/WiFiManager)

closed 2018-01-27 by 1technophile

tobiasfaust, 2018-08-03:
Its possible to integrate WifiManager in ESP32 too?

1technophile, 2018-08-03:
In fact i'm waiting for the integration into wifimanager main repo : 
https://github.com/tzapu/WiFiManager/issues/241

thundergreen, 2018-08-20:
Will we get a gui in the future?