1technophile/OpenMQTTGateway #84: bit confused about setting the protocols

consolacion, 2017-08-10:
I am a bit confused on how to send rf messages from Openhab. Your blog says the following:

```
Switch name "label" {mqtt=">[mosquitto:home/MQTTto433:command:ON:1049666],
>[mosquitto:home/MQTTto433:command:OFF:1049998],
<[mosquitto:home/433toMQTT:command:ON:1049666],
<[mosquitto:home/433toMQTT:command:OFF:1049998]"}
```

However.. what to do if you have two switches with different timings/protocols.

Your github says one has to define the protocol in the topic, like so:

``home/433toMQTT/protocol`,` 

but apparently that is only for incoming RF messages.
Then there are the RFprotocolKey and the RFBitsKey. How/when are those used? I see them defined but not used anywhere and what if you have different protocols??
So suppose I have two switches that both work fine with RCSwitch......how do i call these with your gateway?

1technophile, 2017-08-10:
hello,

> home/433toMQTT/protocol
is for receiving 433mhz protocol data.

So as to send 433mhz with special parameters I advise to take a look to the wiki:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF#send-data-by-mqtt-with-advanced-rf-parameters.

You call your different switches like that:
`mosquitto:home/commands/PLSL_315/433_4/BITS_25:command:OFF:1049998`
`mosquitto:home/commands/PLSL_315/433_2/BITS_25:command:OFF:1049985`

It's not compulsory to use all the parameters.

May I suggest to post questions to the gitter chat.

consolacion, 2017-08-10:
Thanks, I will have a look.
Apologies, I thought this was the place for questions, I have no idea what gitter chat is. Anyway, thanks

1technophile, 2017-08-10:
No problem, you can find the chat there:
https://gitter.im/OpenMQTTGateway/Questions_support

1technophile, 2017-08-17:
Did you succeed with the différent parameters ?

consolacion, 2017-08-18:
I do appreciate your follow up very much. Alas, a busy travel schedule has prevented me from proceeding at this moment, but I will try and then update you. I do not think there will be a problem, seems I was just confused by an example that might have belonged to an older version.

1technophile, 2017-09-27:
Hello, some news about this issue ?

1technophile, 2017-10-06:
I'm closing the issue don't hesitate to reopen it if you need

closed 2017-10-06 by 1technophile