1technophile/OpenMQTTGateway #85: Problem 433toMQTT

zibous, 2017-08-12:
I have a problem with some to use the 433toMQTT, because no MQTT Message will be send. 

Serial Monitor Message:
----------------------
WiFi connected
Setup RF Modul TRANSMITTER
Setup RF Modul RECEIVER
Attempting MQTT connection...
connected to broker
subscription OK to the subjects defined

The Sensor BME280 and DTH22 sends MQTT Messages and 
[image]
works well.

Do I do something wrong with the wiring or is the Areno 433Mhz signal not supported?
If I connect the components directly to the Raspberry, then I get the signals and can process these.

Used Components
433Mhz XY-FST Sender / Transmitter
433Mhz XY-MK-5V Empfänger / Receiver
https://www.amazon.de/gp/product/B071J2Z3YK/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

Arendo - Outdoor Funksteckdosen-Set
https://www.amazon.de/gp/product/B00QXCRI2M/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

1technophile, 2017-08-12:
I would propose two things : 
-try to upload rcswitch examples to an arduino if you have nd check this way the receive nd send functions of the gateway
-could you give what you use in user_config.h as pin definition ?

zibous, 2017-08-13:
Hi Florian,
I connected VCC to VUSB or as it is marked on my PCB: VU. A quick verification 
learned there is no power on the Vin.

[image]

All works no perfect:
``
Receiving RF
isAduplicate
Adv data RFtoMQTT
Sending RFtoMQTT
5471
RF sent by MQTT
Hey I got a callback 
Storing signal
Min index: 11
store code : 5471/1159477
Col: val/timestamp
0:1361/5638
1:5471/7115
2:1361/9184
3:5471/11263
4:1364/12344
5:1361/14724
6:5471/15696
7:1361/29496
8:5471/29665
9:5457/68046
10:1364/87464
11:5471/1159477
Data stored
5471

MQTT
Protocol: 1
Bits:24
Length:299
Payload: 5471
``

Thanks for your help.

1technophile, 2017-08-13:
Glad to hear, could you close the issue in this case

zibous, 2017-08-14:
Thanks for your help.

closed 2017-08-14 by zibous

bachoo786, 2017-08-14:
@zibous can you please explain to me how did you download your codes to the nodemcu? which codes? what did you change in the code? 

thanks

zibous, 2017-08-15:
Hi Florian,

I use the AZDelivery NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E,
without the changing the source code and flashed with ATOM or VisualCode via the serial interface. 
I use an iMac and the USB / CH340 Serial Communication Driver for Mac.

In my case i have only change the 5V on the AZDelivery NodeMCU, so I could use all the components.

[image]

https://www.amazon.de/gp/product/B06Y1ZPNMS/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1