1technophile/OpenMQTTGateway #86: I cannot find the ino file for MQTT to 433 & 433 to MQTT

bachoo786, 2017-08-14:
Hi 

Can you please help?

I cannot find the ino file for MQTT to 433 & 433 to MQTT?

Thanks.

1technophile, 2017-08-14:
Hi the name is ZgatewayRF

closed 2017-08-14 by 1technophile

bachoo786, 2017-08-14:
thanks but what pin shall I connect the transmitter and the receiver?

1technophile, 2017-08-14:
did you check the wiki?
https://github.com/1technophile/OpenMQTTGateway/wiki

You should find all in it

bachoo786, 2017-08-14:
I used your old code (please see below). I am only interested in 433 to MQTT and MQTT to 433. Everything works well except for the transmitter. I have connected it to pin D2 on my NodeMCU and the receiver in pin D1, the receiver works fine but the transmitter I can subscribe to the topic and send out commands and receive them but the rf outlet does not turn on :( 

Please help?!

Thanks.

```
/*
  433toMQTTto433  - ESP8266 program for home automation
   Act as a wifi gateway between your 433mhz nodes and a MQTT broker
   Send and receiving command by MQTT

  This program enables to:
 - receive MQTT data from a topic and send RF 433Mhz signal corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received 433Mhz signal
  Contributor:
  - 1technophile
  Based on:
  - MQTT library (https://github.com/knolleary)
  - RCSwitch (https://github.com/sui77/rc-switch)
  - ESP8266Wifi

  Project home: https://github.com/1technophile/433toMQTTto433_ESP8266
  Blog, tutorial: http://1technophile.blogspot.com/2016/09/433tomqttto433-bidirectional-esp8266.html
Permission is hereby granted, free of charge, to any person obtaining a copy of this software
and associated documentation files (the "Software"), to deal in the Software without restriction,
including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so,
subject to the following conditions:
The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED
TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL
THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF
CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
Some usefull commands to test gateway with mosquitto:
Subscribe to the subject for data receiption from RF signal
mosquitto_sub -t home/433toMQTT
Send data by MQTT to convert it on RF signal
mosquitto_pub -t home/MQTTto433/ -m 1315153
*/
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <RCSwitch.h> // library for controling Radio frequency switch

RCSwitch mySwitch = RCSwitch();

//Do we want to see trace for debugging purposes
#define TRACE 1  // 0= trace off 1 = trace on

// Update these with values suitable for your network.
#define wifi_ssid “*********"
#define wifi_password “*************"
#define mqtt_server “*********"
#define mqtt_user “***********" // not compulsory if you set it uncomment line 127 and comment line 129
#define mqtt_password “*********" // not compulsory if you set it uncomment line 127 and comment line 129

//variables to avoid duplicates
#define time_avoid_duplicate 3000 // if you want to avoid duplicate mqtt message received set this to > 0, the value is the time in milliseconds during which we don't publish duplicates
// array to store previous received RFs codes and their timestamps
long ReceivedRF[10][2] ={{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0}};

#define subjectMQTTtoX "home/commands/#"
//RF MQTT Subjects
#define subject433toMQTT "home/433toMQTT"
#define subjectMQTTto433 "home/MQTTto433/"
 
//adding this to bypass to problem of the arduino builder issue 50
void callback(char*topic, byte* payload,unsigned int length);
WiFiClient espClient;

[... 196 lines cut ...]
    //send received MQTT value by RF signal (example of signal sent data = 5264660)
    mySwitch.send(data, 24);
}

//send MQTT data dataStr to topic topicNameSend
void sendMQTT(String topicNameSend, String dataStr){

      char topicStrSend[26];
      topicNameSend.toCharArray(topicStrSend,26);
      char dataStrSend[200];
      dataStr.toCharArray(dataStrSend,200);
      boolean pubresult = client.publish(topicStrSend,dataStrSend);
      trc("sending ");
      trc(dataStr);
      trc("to ");
      trc(topicNameSend);
}

//trace
void trc(String msg){
  if (TRACE) {
  Serial.println(msg);
  }
}
```