1technophile/OpenMQTTGateway #89: The Wire.requestFrom() does not need Wire.beginTransmission() or Wire.endTransmission().

Koepel, 2017-08-19:
In the file "ZsensorBH1750.ino", the Wire.requestFrom() is encapsulated by Wire.beginTransmission() and Wire.endTransmission(). Those can be removed, since the Wire.beginTransmission() and Wire.endTransmission() should only be used when writing data.

The Wire.requestFrom() does a complete I2C transaction on its own.

If the sensor is not connected or there was a detectable problem on the I2C bus, the Wire.available() will probably return zero. If you want, you could test if the same number of bytes is received as was requested. However, if that fails, there is a serious hardware I2C problem.
The BH1750 itself always returns a value, but if it is read too soon, the value is from a previous measurement.

```
    // Check if reads failed and exit early (to try again).
    Wire.requestFrom(BH1750_i2c_addr, 2);
    if(Wire.available() != 2) {
      trc(F("Failed to read from LightSensor BH1750!"));
    }else{
      i = Wire.read();
      i <<=8;
      i |= Wire.read();  

      // Calculate the Values
```

1technophile, 2017-08-19:
Hello thanks for submitting this,
@hannesdi, would it be possible to take a look to that please?

1technophile, 2017-10-06:
@HannesDi do you have some thoughts on this

closed 2017-11-11 by 1technophile