1technophile/OpenMQTTGateway #90: Query DNS Server before connect

tegila, 2017-08-22:
I've using your code with my homeassistant server to integrate with my 433mhz PIR and door sensors.

But after changing my router 3 times, while searching for a good one, I was left without the integration OpenMQTTGateway provides and have to put back the nodemcu in the computer to write the new IP Address.

After a little research I found some function in the WiFi lib that can fix my problem. 

The best solution will rely on something that check between IP String or Hostname and find the best way to reach the MQTT server.

`// client parameters
PubSubClient client;`

`  trc(F("WiFi connected"));
  IPAddress remote_addr;
  WiFi.hostByName(mqtt_server, remote_addr);
  client = PubSubClient(remote_addr, mqtt_port, callback, eClient);`

1technophile, 2017-08-22:
hello,

What I understand is that 
`WiFi.hostByName(mqtt_server, remote_addr);`
return the IP adress of the host named mqtt_server.
What is the benefit of checking against IP string?

1technophile, 2017-09-12:
@tegila could you detail your proposal benefit?

1technophile, 2017-09-27:
As i haven t feedback i close this issue, reopen it if you need

tegila, 2017-09-27:
Hi @1technophile,

Sorry for missing the feedback (i had muted my notifications from github)

So, yes that is exactly what this function does, it get the ip from the dns name.

The benefit come from the fact that when I change my router, or reset, or anything that can change my realm of IP's from the router. I didn't need to get my OpenMQTTGateway device and reprogram it with the new IP address. 

My homeassistant has always the same domain name ("hass"), so I can always query the dns against this name and never need reprogram my OpenMQTTGateway again.

closed 2018-01-06 by 1technophile