1technophile/OpenMQTTGateway #96: User_config.h:200: error: 'D7' was not declared in this scope

kukudemajia, 2017-08-31:
Arduino:1.8.4 (Windows 7), 开发板："ESPresso Lite 1.0, 80 MHz, 115200, 4M (3M SPIFFS), nodemcu, Disabled, None"

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'int getMin()':

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\OpenMQTTGateway.ino:288:32: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if (ReceivedSignal[i][1] < minimum) {

                                ^

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'boolean isAduplicate(long int)':

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\OpenMQTTGateway.ino:300:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

  if (ReceivedSignal[i][0] == value){

                              ^

In file included from E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\OpenMQTTGateway.ino:42:0:

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\ZgatewayBT.ino: At global scope:

User_config.h:200: error: 'D7' was not declared in this scope

     #define BT_RX D7 //ESP8266 RX connect HM-10 TX

                   ^

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\ZgatewayBT.ino:39:27: note: in expansion of macro 'BT_RX'

 SoftwareSerial softserial(BT_RX, BT_TX);

                           ^

User_config.h:201: error: 'D6' was not declared in this scope

     #define BT_TX D6 //ESP8266 TX connect HM-10 RX

                   ^

E:\OpenMQTTGateway\Arduino\OpenMQTTGateway\ZgatewayBT.ino:39:34: note: in expansion of macro 'BT_TX'

 SoftwareSerial softserial(BT_RX, BT_TX);

                                  ^

exit status 1
'D7' was not declared in this scope

"//  #define ZgatewayBT  " is ok.

1technophile, 2017-08-31:
Hello, Some context informations perhaps  ?

kukudemajia, 2017-08-31:
I see what's wrong.
line: 180
    #define BT_RX D7 //ESP8266 RX connect HM-10 TX
    #define BT_TX D6 //ESP8266 TX connect HM-10 RX
should be:
    #define BT_RX 13 //ESP8266 D7 RX connect HM-10 TX
    #define BT_TX 12 //ESP8266 D6 TX connect HM-10 RX

closed 2017-08-31 by kukudemajia

1technophile, 2017-08-31:
ok I will modify the default settings

kukudemajia, 2017-08-31:
cheer