1technophile/OpenMQTTGateway #100: Xiaomi mi flower sensor

danielpondruff, 2017-09-09:
Hi,

Is it hard to implement this Device and feature in general ? 
My idea is to pair the sensor with openmqtt and read out the values from sensor and push it to mqtt.

Here is a example based on python:
https://github.com/marcelrv/miflora

1technophile, 2017-09-09:
Hi,
You mean to implement xiaomi flora reading with hm10  ?

danielpondruff, 2017-09-10:
Hi,
exactly. is this feature makeable?

1technophile, 2017-09-10:
In theory yes, it s even a good idea

1technophile, 2017-10-06:
Currently trying to make talk the HM10 with the miflora, not so easy. Some ressources about that:
http://www.martyncurrey.com/hm-10-bluetooth-4ble-modules/
https://forum.arduino.cc/index.php?topic=406280.0
I will continue to investigate

1technophile, 2017-10-14:
Thanks to Martin Currey advice, by updating the firmware to V601 I now have access to a new command "AT+DISA?" which returns hex values and identify the mi flora by its name.
It is progressing!

danielpondruff, 2017-10-16:
Sounds great !
Can i help you with testing ?

1technophile, 2017-10-22:
Hello, yes it will help. I can confirm now that it is feasible. I need to rewrite the program to process hexa instead of ascii. I hope to publish something next week

1technophile, 2017-10-26:
It's been a long work but here is a beta integration of mi flora, note that:
* so as to enable mi flora reading you must comment ZgatewayBT_stable into config_BT.h
* you need at least v601 of hm10 firmware

1technophile, 2017-11-03:
Hello, does someone tested this beta integration ?

danielpondruff, 2017-11-03:
Hi Florian,

Give me one or two weeks to test your implementation .

Greetings
Daniel

1technophile, 2017-11-06:
Some explanations of what has been done to integrate this:
https://1technophile.blogspot.fr/2017/11/mi-flora-integration-to-openmqttgateway.html

rvdvoort, 2017-11-23:
I tried the code for the mi-flora integration and it seems to work. Unfortunately the gateway is also reporting the BLE adresses of all other devices closeby. Is it possible to only read the data of a specific mi flora by specifying the BLE mac-adress (and interval between 2 polls).

1technophile, 2017-11-23:
Hello,

You can specify the interval by changing TimeBtw_Read into config_BT.h.
The other mac adresses are published for people presence detection.
Maybe the filter of mac adresses should be placed on the home controller , what do you think ?

1technophile, 2017-11-24:
@rvdvoort I close the issue to release the new version.
Don't hesitate to give your answer to the gitter tchat room

closed 2017-11-24 by 1technophile