1technophile/OpenMQTTGateway #101: add ssl support to connect to mqtt broker

superyarik, 2017-09-10:
now it can only connect via not secure port, so connection is not secure.

superyarik, 2017-09-10:
no, i'm using 0.3.1, and can not connect to ssl port of broker. maybe problem in pubsubclient, maybe there is some settings need to be setted additionally.

1technophile, 2017-09-10:
Coudl you give more inputs, what are the settings you set  ?

superyarik, 2017-09-11:
I use cloudmqtt
https://www.cloudmqtt.com/docs.html

Port 13378 work perfect, but ssl port is notwork, client not connect.

1technophile, 2017-09-12:
ssl seems not supported by pubsubclient:
https://github.com/knolleary/pubsubclient/pull/251

superyarik, 2017-09-13:
yep, another clients(mqtt dashboards for android) work well.

look like this is another part of joke that "S" in "IoT" is Security :)

8666, 2017-10-08:
I think that SSL is possible on ESP but not on the Arduino (there are libraries). Arduino UNO does not have memory for whole SSL magic (https://github.com/256dpi/arduino-mqtt)

1technophile, 2017-10-09:
thanks for the link

Legion2, 2020-07-09:
For TLS we need a the current time as unix epoch, this could be hardcoded which is bad and will require to update the sketch from time to time. The better solution is to use the NTP protocol and let the esp request the time from a time server.

Legion2, 2020-07-09:
I found this library https://github.com/arduino-libraries/NTPClient which can be used.

Legion2, 2020-07-10:
I now have TLS working on ESP8266 with server certificate, but it does not work with the root ca certificate.

@1technophile thanks for sponsoring me 👍

1technophile, 2020-07-10:
> I now have TLS working on ESP8266 with server certificate

Good news! It's already a good step

> @1technophile thanks for sponsoring me 👍

With pleasure, you have made very good contributions to this project

Legion2, 2020-07-10:
Ok it now also works on ESP32 but there only the root ca certificate works and not the server certificate. I will create a PR in the next days.

Legion2, 2020-07-14:
The NTP library is not needed, ESP8266 and ESP32 have them build in and also the CA certificate works now (I don't know why, I did not change anything). Here is the PR #699

commit 9ea25e85 (2020-07-20): added support for secure TLS connections
on ESP32 and ESP8266 close #101
commit f104b5d3 (2020-07-24): added support for secure TLS connections
on ESP32 and ESP8266 close #101
commit 4b42f893 (2020-07-25): added support for secure TLS connections
on ESP32 and ESP8266 close #101
PR #699 (2020-08-02): added support for secure TLS connections
support TLS on ESP32 and ESP8266 close #101

TODO:
- [x] documentation
- [x] tests
closed 2020-08-02 by 1technophile