1technophile/OpenMQTTGateway #102: Restarting of Ras Pi causes the last IR command to be resent

BertrumUK, 2017-09-11:
Have had this issue for a while. If I reboot my Ras Pi (MQTT Broker) my IR transmitter fires the last command even though there is no MQTT command being sent from the broker. I am running the IR transmitter from a NodeMCU.

I am using the IR to control a digibox and use this to power it up in the morning - a reboot causes the digibox to switch off which isn't very popular with the family.

1technophile, 2017-09-12:
Do you have this behaviour even if openmqttgateway is not connected  ?

DenDeze, 2017-09-12:
Are you retaining your messages?

Met vriendelijke groet,

Mathias Everaerts
Servicetechnicus
OBSKSW
mathias.everaerts@engie.com
T +31 88 484 95 50
M +31 6 226 357 32

[Logo]<http://engie-services.nl/>

engie-services.nl<http://www.engie-services.nl>

ENGIE Infra & Mobility B.V.
Albert Heijnweg 1, 1507 EH Zaandam
Postbus 113, 1500 EC Zaandam

Spaar het milieu door deze e-mail en de bijlagen niet af te drukken

ENGIE Mail Disclaimer: http://www.engie.com/disclaimer/disclaimer-nl.html

BertrumUK, 2017-09-12:
If I monitor the broker (mosquito) when I reboot the NodeMCU there is nothing sent from the broker telling the NodeMCU to fire off the last IR command. 

I have been running this for about 3 months and it has always done it. I updated all the code about a month ago without modifying the base code and only updating the settings for classic wiring and wifi etc.

1technophile, 2017-09-12:
This will retain is only for last will testament of the gateway not for commands

1technophile, 2017-09-12:
here is some info a bout retain flag
http://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages

and this chapter:
Delete a retained message

BertrumUK, 2017-09-13:
Thank you for that. Maybe I need a way to send the power command then a false command that does nothing. A restart of the broker or NodeMCU would then only send the false command. 

Can I keep this issue open a while longer in case someone comes up with a better solution?

gammelobst, 2017-09-13:
have you already tried sending a empty message with the retained flag set to your conflicting topic?
I never had this  with OpenMQTTGateway and FHEM, but i do not set the retained flag while sending sth. from FHEM to the broker.
I had only retained messages with the old name after renaming the gateway, but sending an empty message with retained flag let them go away.
What home automation controller software do you use?

BertrumUK, 2017-09-13:
I use Home Assistant on a Raspeberry Pi. The Pi is running mosquito as the broker. The retain flag is set to “false” on the code running on the NodeMCU and the command on Home Assistant. 

I am running a 433mhz transmitter and a DHT22 temp/humidity sensor from the same NodeMCU. 

If I reboot the Pi or the NodeMCU the last IR command is resent but the 433mhz rf signal does not this is what I don’t understand- why does the last IR command get sent when retain is set to false ?

1technophile, 2017-09-13:
> what I don’t understand- why does the last IR command get sent when retain is set to false ?

the retain flag parameter in the user_config.h is not for commands. It is for last will testament (online/offline detection of the gateway)

Try sending an empty message to the problematic topic or maybe modify your topic as an ultimate solution

BertrumUK, 2017-09-15:
Its not very elegant but I have incorporated a fake IR command to be sent when the power to my digibox turns on. If the broker or the NodeMCU are restarted the fake command gets resent but there is no action taken by the digibox. As I said it's not elegant but it gets around the issue that I am having with the MQTT Switch component of Home Assistant.

Closing this issue. Thanks for all that helped me.

closed 2017-09-15 by BertrumUK