1technophile/OpenMQTTGateway #110: How to use BLE Tracker with Home Assistant?

danielpondruff, 2017-09-18:
I'm currently not quite as clear as I am integrating the BLE Tracker into Home assistant. 
Can someone help me?

The HM10 works and detected devices are also pushed into MQTT

yzg011, 2017-09-19:
this is also my question

1technophile, 2017-09-19:
@petricaM do you think you could write a wiki page about this as you integrated it to your home automation ?

PetricaM, 2017-09-20:
Yes, off course. Although I wouldn't know were to begin the wiki page :) as I haven't written one before.

I think I could post the entire configuration of HA after some housecleaning (only to remove personal information; I'm not a software engineer, thus the code itself for the sensors&scripts&automation could be quite clumsy but it works). I managed to integrate a relatively large number of different standard technologies (RF such as 433mhz, 2.4ghz, Z-wave), products (PIR sensors, MiLight bulbs, Air Conditioner, TV, etc) that speak together either directly to HA or through several gateways. 
And also a not so pretty but quite handy FloorPlan integration :)

PetricaM, 2017-09-20:
Until I have time to place it all together:

Simple usage (to see it in the frontend) would involve setting only of a MQTT sensor (A). The issue is, however, that the sensor would show unknown value if not updated or the beacon is associated (thus the need for B). C came into being as a need to take into account the movement of the BT beacon from/towards the Gateway (meaning that when strength of the signal increases the sensor set at E will be on).
Either D or E binary sensors could be used in automation (however in my example the automation with D binary sensor it would be with "off" rather than "on" for the trigger payload). 

There are several limitations to using the BT presence sensor:
- the BT device needs to be un-associated (so if using a BT token associated to a phone it won't work); the specific case below is for using a Tile associated to my secondary phone on which I keep, most of the time, the BT connection off; when I need to find my keys I open BT connection on the phone
-  due to the fact that Gateway readings are made, in the standard sketch, at 10 seconds (that is the Gateway will scan all BT signals received at each 10s), this is not suitable for using to trigger the lights.
Although readings interval can be shorten and it could be used with MQTT Room Presence component (it requires a BT Gateway in each room monitoring is needed) it is not recomendable to use it for turning the lights on an actual production system. This is because, in practice, a light on would require a less than one second interval between trigger and action in order to be usable. 
I use it to open hallway lights when arriving at home (automation based on the binary sensor E) and it works as it usually takes several seconds from the moment the gateway senses the token until opening the front door. Anyway, redundancy through linking it with other components (such as PIR or open door sensors) would ensure reliability.
I have each group of sensor in a separate file in the sensor/binary_sensor/automation folders so use proper indentation if placing all in "configuration.yaml". Splitting the configuration in multiple files is highly recommended.

I. Sensors:
#A. Setting up initial sensor
```
- platform: mqtt
  state_topic: 'home/BTtoMQTT/AAAAAAAAAAAA' # if the default MQTT topic was updated then this need to be reflected here; replace AA... with id of your device
  name: 'keys_car_state' 
  expire_after: 300 # modify this to desired value (measurement is in seconds)
```
#B. Setting a sensor through altering the initial sensor readings in order to deal with "unknown" state
```
- platform: template
  sensors:
    keys_car: #unique names; this is different from the one above
      value_template: >
        {% if is_state('sensor.keys_car_state', 'unknown') %}
          -100 # RSSI values are negative; an even smaller value could be used 
        {% else %}
          {{states.sensor.keys_car_state.state | float}} #if the name of the sensor from step A was changed, include it here
        {% endif %}

```
#C. Setting up a statistics sensor 
```
- platform: statistics
  entity_id: 'sensor.keys_car' # this is the name of the sensor from step B
  name: 'keys_car_stats' #again, different id
  sampling_size: 10 # the default sample size is 20; can be modified to one's desires; 
```

II. Binary sensors
#D. Setting a binary sensor based on the step A; this is mandatory for usage with more complex automation (such as "for" clause for trigger)
```
- platform: template
  sensors:
    bt_car_keys:
      friendly_name: 'bt_car_keys'
      value_template: >
        {% if is_state('sensor.keys_car_state', 'unknown') %} # modify here the value from step A
          true
        {% else %}
          false
        {% endif %}
```
#E. Setting a binary sensor 
- platform: template
  sensors:
      bt_car_keys_threshold:
        friendly_name: 'bt_car_keys_threshold'
        value_template: '{{ states.sensor.keys_car_stats_mean.attributes["average_change"] | float > 0 }}' #"key_car_stats" is the name from C; mean is used but could be also other statistics (median, change, variation, count, etc)

III. Automation
```
- alias: 'bt_hallway_lights'
  initial_state: 'on'
  trigger:
    - platform: state
      entity_id: binary_sensor.bt_car_keys_threshold # the same name as step E
      to: 'on' 
# with the binary sensor a "for" clause can be used (see below and remove the # if needed); for regular sensors this is not possible; this depends if you live on a upper floor and want to trigger the lights only when in front of the door 
      #for: 
          #seconds: 3
  condition:
    condition: and #a simple condition could be used however I have also a condition that uses different light collors based on a switch; using from the beginning the "condition: and" would be easier to modfiy subsequently
    conditions:
      - condition: state
        entity_id: sun.sun
        state: 'below_horizon'
  action:
    - service: homeassistant.turn_on
      entity_id: light.hallway #replace with your light
```

IV. Hide all from view (after ensuring all works as intended) by including this in the 
```
    "sensor.keys*":
      hidden: true
    "binary_sensor.bt*":
      hidden: true
    "automation.bt*":
      hidden: true
```

1technophile, 2017-09-20:
@PetricaM this is already great, thanks

1technophile, 2017-09-20:
It's here:
https://github.com/1technophile/OpenMQTTGateway/wiki/Home-assistant-integration
you can modify it as you wish

PetricaM, 2017-09-20:
Hi,

I have also modified the formula for real feel with the sensors mentioned by BertrumUK on the same page (I haven’t modified his code as it’s not my initial contribution). 

In Bruh latest sketch he used the real feel computation on the NodeMCU code however it wasn’t really nice because (1) his formula was for F not C as he has a portion of code where it mentions to make the modification for temperature scale reporting but it doesn’t have 2 different formulas for F and C real feel (I haven’t bother to look where the error is and I’ve set the formula from scratch) and (2) NodeMCU’s with his code fire continuously on the MQTT broker (I mean, even without the PIR sensor there’s a message at each 4 or 5 seconds triggered by either a modification in the luminance or temperature or the humidity which I consider excesive).

However, I cannot possibly understand why, in my case, it won’t work to set measurement unit without Â in from of °C. I guess that if I would have been using Windows as base for HA it would be due to locale settings however on the Rpi I haven’t used raspi-config as far as I recall. I don’t know if it works without Â for the other users.

I’m an accountant thus I don’t even know if there’s any difference between C, Python or Java whatsoever 😃 

Regards

- platform: template
  sensors:
    mqtt_real_feel:
      friendly_name: "mqtt real feel"
      value_template: '{{ (((-42.379 + (2.04901523*(((states.mqtt_rf_temperature.state|float)*1.8)+32)) + (10.14333127*(states.mqtt_rf_humidity.state|float)) - (0.22475541*(((states.mqtt_rf_temperature.state|float)*1.8)+32)*(states.mqtt_rf_humidity.state|float)) - (6.83783 * (10**(-3))*((((states.mqtt_rf_temperature.state|float)*1.8)+32)**2)) - (5.481717 * (10**(-2))*((states.mqtt_rf_humidity.state|float)**2)) + (1.22874 * (10**(-3))*((((states.mqtt_rf_temperature.state|float)*1.8)+32)**2*(states.mqtt_rf_humidity.state|float))) + (8.5282 * (10**(-4))*((((states.mqtt_rf_temperature.state|float)*1.8)+32)*(states.mqtt_rf_humidity.state|float)**2)) - (1.99 * (10**(-6))*((((states.mqtt_rf_temperature.state|float)*1.8)+32)**2)*((states.mqtt_rf_humidity.state|float)**2)))-32)/1.8)|round(1) }}'
      unit_of_measurement: 'Â°C'

From: Florian
Sent: Wednesday, September 20, 2017 10:01 PM
To: 1technophile/OpenMQTTGateway
Cc: PetricaM; Mention
Subject: Re: [1technophile/OpenMQTTGateway] How to use BLE Tracker with HomeAssistant? (#110)

It's here:
https://github.com/1technophile/OpenMQTTGateway/wiki/Home-assistant-integration
you can modify it as you wish

1technophile, 2017-09-22:
as the wiki is now completed I close this issue

closed 2017-09-22 by 1technophile

anarchking, 2021-01-18:
Having some trouble publishing topic and payload through home assistant publish a packet tool in mqtt integration. Im trying to update ble interval to 1 sec. From my understanding the topic would be home/OpenMQTTGateway_ESP32_BLE/commands/BTtoMQTT/config and in the payload box '{"interval":0}' or {"interval":0}. I subscribed to the topic home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/xx:xx:xx:xx:xx but the messages only come in once a min. Ive tryed alot of different variations but nothing is working for me. Im not sure what im doing wrong

1technophile, 2021-01-18:
> home/OpenMQTTGateway_ESP32_BLE/commands/BTtoMQTT/config and in the payload box '{"interval":0}' or {"interval":0}

If you do this you are going to [force a scan](https://docs.openmqttgateway.com/use/ble.html#setting-the-time-between-ble-scans-and-force-a-scan)

You should set the interval to 1 instead.

What is the use case behind?

anarchking, 2021-01-18:
Yeah sorry typo I tried '{"interval":1000}' for a sec and '{"interval":0}' for an instant response but was still only seeing them every minute with subscribe

anarchking, 2021-01-19:
Using with mi band 5 for presence detection and other stats.

anarchking, 2021-01-19:
Ha sorry typo again :) figured it out, i had BTtoMQTT instead of MQTTtoBT thanks for the fast response and great project.

1technophile, 2021-01-19:
Glad you got it working