1technophile/OpenMQTTGateway #112: MQTT disconnect after a certain time with HM10

1technophile, 2017-09-19:
I have an issue on the NodeMCU boards I have: on both v0.4 and 0.3 bluetooth module (non-genuine HM10) and (occasionally) RF receiver (superheterodyne type) the MQTT connection gets reset. This does not trigger full reboot of NodeMCU but only disconnects the MQTT connection and reconnects and it takes a few seconds (3 - 4 at max) to reconnect.

I wouldn’t have observed this if not for the board being connected to an IR blaster with an AC unit. When the NodeMCU gets reset, it loads the code for the MQTT switch (it has retain flag) and sends the IR code to the AC (and AC has a buzzer).

I think there was someone else with a similar problem with the bluetooth module.

This does not occur on the Arduino Mega (v.30) but only on the NodeMCU. It might be a defective board problem (I’m using generic boards and they also have wrong prints for D1 pin on both D1 and D2) however it’s not a sampling aspect. I have tested on 3 units out of the 10 units batch and they all get reset.

BertrumUK, 2017-09-19:
Not exactly similar to #102 but with older code I used to get a similar issue with the last MQTT IR command being sent after approx 2 days since the last reboot of the NodeMCU. I updated to your newer code, which added the classic wiring option, and it seems to have stopped. I do however still have the last IR command being sent on a reboot of the NodeMCU or the PI.

PetricaM, 2017-09-22:
I received a (I think) genuine HM10 and there is no issue with this one (24h and counting) while with the clone the issues appeared almost immediately.
I think it's safe to say that the problems are in regard of the clones.

PetricaM, 2017-09-22:
@BertrumUK 
If you used the switch for the IR with the Retain flag is normal to have the NodeMCU sending the code each time it reboots (or the MQTT broker).

BertrumUK, 2017-09-25:
@PetricaM Ok. It is less of a problem now as I issues a fake IR command if I need to restart the broker or NodeMCU. This command is fired but does nothing.

1technophile, 2017-09-26:
hello @PetricaM after 4 days have you seen the issue?

PetricaM, 2017-09-27:
Hi,

I haven't seen it with the original HM10, however I'm away since Saturday. I'll have another look at the MQTT log during this weekend.

Thanks

1technophile, 2017-09-29:
I m going to bring some code modifications for hm10 to try if we can remove this bug by giving more time to the wifi stack of the esp8266

1technophile, 2017-09-30:
@PetricaM @jxjhheric @BertrumUK could you test the last version to see if it corrects your issue

BertrumUK, 2017-09-30:
I now know that the retain flag is causing my issue - confirm that the last IR & RF command gets resent. Not a problem with the RF as there are distinct “on” & “off” commands whereas the IR just toggles the same command. 

Will try the new update when I get time 👍

1technophile, 2017-09-30:
ok maybe some misunderstanding I think you had also the issue with reboot when HM10 is connected, the correction I made is not related to the retain flag but it is for HM10 causing reboot :-(

BertrumUK, 2017-09-30:
Ok, thought so. As I have said previously, now I know what the cause is I have a way around it. Thanks.

1technophile, 2017-11-11:
@PetricaM hello, could you test the new version of ZgatewayBT to see if it reboot ?

* you must comment ZgatewayBT_stable into config_BT.h
* you need at least v601 of hm10 firmware

PetricaM, 2017-11-19:
Hi Florian,

I've tested the sketch with the latest settings (HM-10 v.603 and BT stable commented for mi flora) however it still keeps resetting the MQTT connection. I've setup statistic sensors in HA to track MQTT topic for each NodeMCU and for the new sketch with BT gateway only there are still 3-4 MQTT connection restarts per day (the NodeMCU does not restarts). With BT Gateway and RF Gateway there are about 10-15 restarts per day. With BT Gateway and ADC sensor there are about 40-60 restarts per day. 

For NodeMCU using v0.4 there was no restart for a whole week.

I have some comments for the other components:

1. Can you add an option to completly turn off the builtin led?

2. Is there any way to increase the IR raw message size? Or to simply forward the raw signal received (I don't have one but I think this is the Broadlink RM functionality)?

I have a AC using one of the Japanese protocols with length of 112bits. At the moment - with the publish unknown protocol option set - the IR message read by OpenMQTT is incomplete (412 character long however it should have been 942). I'm unsure whether the issue resides with IRREmote or OpenMQTT (I've modified pubsub to 1,024 and sending of these 900+ char messages to IR blaster works however IR receiver doesn't work with long payloads). 

I tried during the summer to follow the instructions set a new protocol for IRRemote (I managed to identify the protocol requirements) however the autumn came and I've set asside this project :).  

If needed, I can help (after some 2-3 weeks when I get home again) as I have an AnalysIR license and I can decode the IR signals.

3. Can you advise on what settings should be made to ADC sketch in order to have regular readings (regardless of threshold) sent to MQTT broker? 

I've tried connecting a sound sensor to OpenMQTT and with the default 50 threshold the noise need to be really high to be picked up. Using a lower than 10 threshold the ADC sensor keeps firing each second so I think that for the sound sensor the reading should be sent at about 30-60 seconds regardless of the threshold.

Regards,

1technophile, 2017-11-22:
Hello @petricam

Thanks for this feedback. Regarding your questions.
1. What kind of board do you have for my information ? 

2. I have to do some test and indeed i could implement a strictly identical raw forwarding. Could you open a different issue for that ? 

3. So as to have regular readings you have to increase the value of TimeBetweenReadingADC into config_adc to the time needed in ms.
Moreover I think this time is currently to low by default. I will increase it to at least 200ms and add. Some code on ZsensorADC to let more space to the esp8266 so as to do its stuff.
Regarding the other points

PetricaM, 2017-11-22:
Hi Florian,

I've added a request for IR raw forward.

I'm using several different setups:
- Arduino Mega and Ethernet Shield for RF Gateway only (running a very old version of OpenMQTT, I think 0.3 or even older :) );
- NodeMCU v3 (at least that's what they are saying but there isn't an official v3 iteration) with DHT22 and LDR sensor (ADC) in each room; for two of these units I've also placed an IR blaster and I've attached them to AC units (only On and Off IR Raw payloads are recorded) and set with Home Assistant's Generic Thermostat;
- NodeMCU v3 with IR blaster for controlling different devices (receiver, Xbox); I'm planning to add an IR receiver to one NodeMCU to receive IR raw payloads and forward them to other devices;
- NodeMCU v3 with BLE gateway but meanwhile I've added also a RF and an ADC and the MQTT restarts kept firing.

1technophile, 2018-02-06:
https://github.com/esp8266/Arduino/issues/1426

1technophile, 2020-06-07:
I'm closing this issue, if you still have comments to add, feel free to reopen it.

The BLE gateway with HM10 has been refactored for info on V0.9.4, you may take a look to it and repopen the issue if it reappears again.

closed 2020-06-07 by 1technophile