1technophile/OpenMQTTGateway #117: How to determine state of multiple RF devices?

jjlawren, 2017-09-24:
Unless I’m using things incorrectly, it appears that all RF devices share the same MQTT state topic. Ideally the payload could be retained so actual switch states could be preserved if this device was restarted.

gammelobst, 2017-09-24:
This sounds like a job for your HA-Controller and not the gateway. MQTT just transports the message, configure the state in HA-Controller and let it change if a  new command is recieved. There is no need to Retain the payload, but it could be resend on restart of the gateway...

jjlawren, 2017-09-24:
Restart of the HA software is also a concern. If state is held by the MQTT broker, retaining per-device state would reflect the true state as RF devices can’t be polled.

gammelobst, 2017-09-25:
U could use you HA Controller to set a corresponding retained topic like home/switches/coffemachine/state to "on" or "off" just like the state changes. 
OpenMQTTGateway is just for sending and recieving, but not for storing states, which would require some kind of array to store the corresponding codes to on and off states. Sounds like wasting memory, adding not really needed complexity because you already have needed info in your HA-controller. Just try sending it from there pls

1technophile, 2017-09-26:
I think @jjlawren is more thinking about retaining the state at broker level?
Ho w do you see the link between the two states of one wall plug that has two different RF codes?

jjlawren, 2017-09-26:
That's correct. Neither the OpenMQTTGateway device nor the HA software should try to retain the state of the RF devices when MQTT itself provides a mechanism via retaining. However, a unique topic which publishes the current state would be needed for each controlled RF device.

If you're familiar with Sonoff-Tasmota, they have an overview of their MQTT design here which allows for this type of setup: https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

gammelobst, 2017-09-26:
ok, but the Sonoff are wireless switching relays, i think. It is no problem for a switch to send his state to the broker as it switches itself and can detect it's state. OpenMQTTGateway is a gateway as the name states and does not know if the command has switched anything. And it doesn't know which switch and state is corresponding to a decimal code(at least not without modifing). The best thing is to send 2 topics to your broker - one to switch the whatever and a retained "on" or "off" to the state topic of your switch.
In here we need to link 2 decimal values to a switch, define the switch first and set or learn those codes to on and off - so we could send a retained "on" to home/OpenMQTTGateway/GardenLightFrontLeft/state to switch the light. With the risk of resending the codes which would probably make probs with a switch that can just toggle (same code for on and off)
But i still think this would add unneeded complexity as we just want to send and recieve commands. HA controller software is needed to unlock the full potential of the gateway...

1technophile, 2017-09-27:
I think current implementation of rcswitch and irremote library indeed doesn't enable to do that easily at the broker level. As said due to the fact that the couple gateway/broker doesn't know that the code A will send an ON command for a device 1 and a code B will send an OFF command to the device 1.

gammelobst, 2017-09-27:
But if we implement a relay that is switched with an output pin - for this it would be an easy thing to do.
Or if the gateway senses the state via an INA226 it would also be easy.

1technophile, 2017-09-27:
I agree for sensors it is relevant

MrPasty, 2017-10-03:
being able to query the state of a device would be very useful, not sure how you could achieve this by not using MQTT ?

1technophile, 2017-10-03:
we could do this with the sensors or with rfm69 but not possible with off the shelf cheap 433mhz sensors

jjlawren, 2017-10-03:
That's kinda my point, though. It's a best effort attempt at persisting the state. Both the HA controller _and_ OpenMQTTGateway can rely on MQTT to restore the last known state if a reboot or disconnection at either end occurs.

8666, 2017-11-20:
The broker will retain the topic depending on the way it is published. Then it will resend the states when.the HA subscribe. I use the lwt topic to monitor the status of this gateway, but when I restart the HA software I do not get the mesaage when resubscribe. I don't think it is wise to have this logic for any other topic.

1technophile, 2017-11-20:
When you restart HA you don't get the gateway state ?

8666, 2017-11-20:
Hmmm with mqtt-spy I get the status. With HA it does not reload. Then it is not related to the RF gateway..will fix my problem later

1technophile, 2017-11-25:
At this time I prefer not putting on gateway level a storage of signals values apart from duplicate removal

closed 2017-11-25 by 1technophile