1technophile/OpenMQTTGateway #118: OpenMQTTGateway dont need RF2 to work with KAKU

animavitis, 2017-09-24:
Hi,

There is simple way to use KAKU (transmit & recive) in OpenMQTTGateway, simply change rc-switch liblaryto this branch:

https://github.com/max-weller/rc-switch/tree/intertechno-receive

Tested with my KAKU switches & NodeMCU

1technophile, 2017-09-24:
Hello,

Thanks for the tip!
@dendeze could you test if it is working with your devices?

animavitis, 2017-09-25:
after more testing i must tell that this version of rc-switch is not stable :/ 
After couple of hours my nodeMCU throws some exceptions & restart itself randomly, but still it can be used to read codes from KAKU remote.

1technophile, 2017-09-26:
do you have some inputs about these instabilities?
is the nodemcu that restart or do you have other symptons?

animavitis, 2017-09-28:
NodeMCU restart itself randomly, usually when i try to use RF transmitter couple of times in short period of time. I will try to debug it

animavitis, 2017-10-02:
I was testing it and same restart can happens on NewRemoteSwitch & rc-switch#intertechno-receive. 
I dont know how to debug root cause of restarts, so we can close this issue.

thx for help

closed 2017-10-09 by 1technophile