1technophile/OpenMQTTGateway #120: sofeware not support new irremoteESP8266 library

yzg011, 2017-09-25:
构建选项已变更，全部重新构建
C:\Users\Administrator\Desktop\RasWxMusicbox-master\OpenMQTTGateway\ZgatewayIR.ino: In function 'boolean IRtoMQTT()':

ZgatewayIR:66: error: 'USECPERTICK' was not declared in this scope

       rawCode = rawCode + (results.rawbuf[i] * USECPERTICK);

                                                ^

exit status 1
'USECPERTICK' was not declared in this scope

1technophile, 2017-09-25:
Please be more detailled, which version of the gateway which version of irremote esp8266 ? 
In all the case you should use the libraries linked into the libraries directory.

yzg011, 2017-09-25:
the last new version of gateway，irremote esp8266 version 2.1.1 ok
irremote esp8266 version 2.2.0 not work。

gammelobst, 2017-09-25:
you have to copy the supplied library to your system arduino lib folder.
I got the same error when using another than the supplied version, but after copying IRRemoteESP8266 to the lib folder it compiles and work like a charm.
cya

1technophile, 2017-09-26:
@gammelobst are you able to compile with the last version of Irremote_esp8266?

gammelobst, 2017-09-27:
i can compile with the here mentioned and linked lib.
But if i update to the latest version via lib-manager than i get the same error.

1technophile, 2017-09-27:
Ok thanks for the feedback i will include the last version on v0.5

closed 2017-10-05 by 1technophile