1technophile/OpenMQTTGateway #131: Implement HDMI CEC <-> MQTT

1technophile, 2017-10-09:
so as to interface HDMI CEC compatible devices (TV, disc reader...)

https://code.google.com/archive/p/cec-arduino/

8666, 2017-11-20:
This is easily done with Rasperry Pi as the hdmi card support it. Other way is to use hdmi-usb adapter. Do not see the point for including hdmi here.

1technophile, 2017-11-20:
Everybody doesn't have necessarily a pi connected to the tv. 
In term of architectire the point is to let to microcontrollers low level functions like this kind of communications.
This way you don't mess up with binding/addons/plugins in your controller hardware and just to have it connected to mqtt and put the configuration files
But as the others requests it will be prioritized following users votes.

8666, 2017-11-20:
Sniffing might be easy but for sending commands you need to emulate full HDMI device. I have some experience with libcec library. You actually need to make something like this but with mqtt/lan not with USB: https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter#

1technophile, 2017-11-20:
Thanks for the info

8666, 2018-01-16:
There is an elegant solution for this, NodeRed on the Pi with HDMI CEC module. Then you can publish them on MQTT.

https://www.npmjs.com/package/node-red-contrib-cec

closed 2018-05-12 by 1technophile